emsec/ChameleonMini #10: A problem about the "unknown device"

wyatuestc, 2015-08-18:
I made a ChameleonMini by myself.
Here is my step to download the firmware:
1. connect the board to PC
2. use avr isp mkii to download the bootloader(RevE-atxmega32a4u_104_modified.hex)
[image]

I can see a device named ATxmega32A4U
[image]

3.after that, I dowload the firmware/eeprom Chameleon-Mini.hex/Chameleon-Mini.elf with avr isp mkii,and I got this:"Unknown device", and I also use the FLIP to download the firmware/eeprom, I got the same result.

[image]

[image]

I want to know what happened.

Tanks :)

wyatuestc, 2015-08-18:
There must be something wrong.
When I open VMware and run a virtual machine(ubuntu 14.04),  I found the hardware is working, I test it many times.....
It's wired....
I guess there is something wrong in the USB driver......

[image]

[image]

wyatuestc, 2015-08-19:
I think I solve problem when I used the newest firmware....
There are something promblems on my hardware.....

david-oswald, 2015-08-19:
Yes, you for sure need the newest FW - the older one contained a bug that caused the Chameleon to fail with newer Flash chips (actually this error should have also affected older Flash, but it seemed more tolerant). Anyway, this is fixed in the new FW.

Re VMWare: Have not tried, any reason why you need this? Also, wasn't VMware working before (and now stopped)? Strange bug, if you find out anything pls let us know.

Re command line: Have you seen this page: https://htmlpreview.github.io/?https://raw.githubusercontent.com/emsec/ChameleonMini/master/Doc/Doxygen/html/_page__command_line.html

closed 2015-08-21 by wyatuestc

theforcedk, 2015-09-16:
In order to use USB's in virtualized environments, you usually have to add filters to ensure that the USB is grabbed by the guest OS before the host OS when plugged in, otherwise you'll get stability issues and the USB will connect/disconnect/connect/disconnect. For VirtualBox, this works with ChameleonMini: http://www.atmel.com/webdoc/atmelstudio/atmelstudio.faq.virtualbox.html