emsec/ChameleonMini #12: Support for ISO15693

niklaus520, 2016-09-27:
Hi guys, I was wondering weather the ChameleonMini supporting ISO15693 currently? Cause now I'm trying to emulate the TI Tag-it HF-I Plus Inlay, and I made ChameleonMini by my own once, the ChameleonMini wiki says it can emulate ISO15693. But I didn't find the correct instructions and manual to change ChameleonMini to ISO15693 mode. Besides, I read the ChameleonMini source code, it only contains ISO14443 protocol. 

So here is the question, can ChameleonMini emulate ISO15693 card now? If not, do you guys plan to add the function as you described in your wiki? And I really love your awesome job. 

Best Regards!

dxls168, 2016-09-28:
VCD to VICC
Supports two amplitude modulation 100% ASK and 10% ASK
 VICC to VCD
Use subcarrier modulation mode, dual-support FSK and ASK subcarrier debug mode single subcarrier two types of modulation.
ChameleonMini RF circuit should be supported.

david-oswald, 2016-09-28:
Yes, technically it should be possible to emulate ISO15693. It will require writing an according codec and the tag application. We will consider adding this in a future FW release. If you want to implement this yourself, I would suggest waiting until RevG with the new firmware is released, since there are changes in the HW and SW architecture. We will not implement further FW updates for RevE.

dev-zzo, 2016-10-12:
Any chance of getting a guide on how to write a codecs and applications? Reading the fine code might delay things quite a bit.

david-oswald, 2016-10-14:
Created a separate issue #33 for the last question.

DrWummi, 2016-10-27:
is there some active ISO15693 development i could join, or should i just go ahead and start spinning my own?

pruwait, 2016-11-28:
Need emulate ICODE 2k 2048bit
1. UID
2. IC Version
3. AFI
4. DSFID
5 64 block data with 32 bit

pruwait, 2016-11-28:
[image]

geo-rg, 2016-11-28:
@DrWummi Now there has some time passed and it seems like we have forgotten to answer your question. Sorry for that. :/ The answer is: No, there is no development on ISO 15693 support currently. But based on the reactions to this thread, I think everyone would appreciate if you would start spinning your own ideas on an ISO15693 codec and let everyone benefit from it. I also think that you won't be alone so long with developing a ISO15693 codec.

geo-rg, 2016-11-28:
@pruwait Since there is no ISO15693 support right now, there also is no ICODE support. Only after the first one is realized, an ICODE emulation support is possible.

DrWummi, 2016-11-28:
@geo-rg 
yes im already working on it - i will put it on github as soon as it takes form

sadly i got not so much time at the moment, and understanding (possibly fixing) #83 is my main concern at tne moment.

if anyone would like to join me in the ISO15693 codec development feel free to contact me via mail => rfid (ä) wummi.at

dev-zzo, 2016-12-23:
I'd like to join as soon as DESFire code is more or less completed. :-)

alexrestart35, 2017-01-18:
David-oswald, DrWummi, I am a new user. Please ask to clarify. 
ChameleonMini supports ISO 15693 emulation ICODE, now ??

geo-rg, 2017-01-18:
@alexrestart35 No, unfortunately ISO15693 support is not realized right now.

m2otech, 2017-03-28:
Any news on ISO15693 support? I currently own a Proxmark3 and I am trying to understand/improve ISO15693 support there. Is anyone familiar with both devices and can recommend which one (Proxmark or ChameleonMini) is more suitable / has better chances of being able to emulate ISO15693 tags?

geo-rg, 2017-04-05:
@m2otech AFAIK the ISO15693 codec is under development but not yet ready.

m2otech, 2017-04-05:
Thanks @geo-rg Any expected date of release? Is it only a software thing (i.e. if I buy a Chameleon now I can update it later to the new codec)?

alexrestart35, 2017-04-13:
Thank @geo-rg, very expect codec. I also has a Proxmark device. It is able to work with 15693.
@geo-rg, what can I do to help? I can run some tests or anything else.

david-oswald, 2017-04-16:
Yes the codec is a software thing, so the current HW will work.

HacKanCuBa, 2017-05-19:
I would like this support, too!

m2otech, 2017-07-04:
@geo-rg (or anyone) could you point me to the right direction in order to develop a codec for ISO15693? Where in the code should I start looking? Is there work done that I can improve? Can I use another codec as "template"?

geo-rg, 2017-08-31:
Hi @m2otech @HacKanCuBa @alexrestart35 @dev-zzo @DrWummi @pruwait @r-ising @niklaus520 @dxls168 
please check out #136. In this pull request you can also find a short list of how you can contribute to improve this first draft of the ISO 15693 codec.

OwenCalvin, 2018-12-28:
Hello any progress to emulate an ISO 15693 tag ?

fptrs, 2019-08-19:
see #182

closed 2019-08-19 by fptrs