emsec/ChameleonMini #13: A strange problem

dxls168, 2016-09-28:
Hi guys:
             I made a ChameleonMini recently.When I use a card reader like ACR122 or RC522 to read, it performed well.
              but when I use a card reader like Mobile phone or access control,it could not pass the second auth.
              I Found that When I send NT, it did not respond.I think it was I missed the response time.But the first auth with no problem.
            Looking forward to your reply.
                                                                                                                                 dxls

david-oswald, 2016-09-28:
This could be a problem with transmission errors due to smaller antennas / "weaker" readers. In RevE, first consider fully removing the "passive power" section ie. the bridge rectifier diodes directly at the antenna. 

If you think this is a timing problem, did you compare to an original MFClassic using a scope?

dxls168, 2016-09-28:
hello david:
                   Very grateful for your reply! 
                    I tried to get rid of D2 and D4 and R5 but the problem remains the same, I'm finally ready to try XMEA128A4, hoping to improve.Look forward to your new open source products！
                                                                                                                                            dxls

david-oswald, 2016-10-08:
This problem should be solved with RevG. If not, please open a new issue.

closed 2016-10-08 by david-oswald

feifeilanniao, 2016-11-13:
My chameleonmini rev_G have the same problem.When i use a Mobile phone  to read,Authentication fails

dxls168, 2016-11-14:
I think the problem may have originated from the program. @feifeilanniao

feifeilanniao, 2016-11-14:
may be.