emsec/ChameleonMini #16: chamtool.py errors while executing CONFIG=?

hsanjuan, 2016-10-08:
```
$> python3 chamtool.py -p /dev/ttyACM0 -v
[2016-10-08 13:38:47.531460] Opening serial port /dev/ttyACM0 succeeded
[2016-10-08 13:38:47.532812] Executing <VERSION?>: 101:OK WITH TEXT
[2016-10-08 13:38:47.533115] Response: ChameleonMini RevG 161008 using LUFA 151115 compiled with AVR-GCC 4.8.3
[2016-10-08 13:38:47.533382] Executing <CONFIG=?>: Based on the open-source NFC tool ChameleonMini
Traceback (most recent call last):
  File "chamtool.py", line 205, in <module>
    main()
  File "chamtool.py", line 168, in main
    if (chameleon.connect(args.port)):
  File "/multimedia/git/ChameleonMini/Software/Chameleon/Device.py", line 96, in connect
    result = self.getCmdSuggestions(self.COMMAND_CONFIG)
  File "/multimedia/git/ChameleonMini/Software/Chameleon/Device.py", line 168, in getCmdSuggestions
    result = self.getSetCmd(cmd, self.SUGGEST_CHAR)
  File "/multimedia/git/ChameleonMini/Software/Chameleon/Device.py", line 165, in getSetCmd
    return self.writeCmd("{}{}{}".format(cmd, self.SET_CHAR, arg))
  File "/multimedia/git/ChameleonMini/Software/Chameleon/Device.py", line 134, in writeCmd
    statusCode, statusText = status.split(":")
ValueError: need more than 1 value to unpack
```

hsanjuan, 2016-10-08:
It's because VERSION? gives a multiline response and only the first line of it is processed

timokasper, 2016-10-08:
ah thanks for this feedback, we have added 3 lines just yesterday so hasn't been tested yet --> we will fix it.
 
TK

geo-rg, 2016-10-24:
Please verify if #25 - which is merged now - solves the problem for you.

hsanjuan, 2016-10-31:
It does solve the problem. THANKS!

closed 2016-10-31 by hsanjuan