emsec/ChameleonMini #17: Windows 10 can't find he ChameleonMini

PetersPGM, 2016-10-08:
I received my ChameleonMini today. Great!

I checked the Getting Started to learn I had to upgrade the firmware. I downloaded the files and hit the first bump. It needed libusb0.dll. After getting and installing that, I continued.

Pressing the RBUTTON when connecting the device to the PC does not show anything on the device. Is that correct? I expected some indication the device is in bootloader mode.
Connecting the device to the computer without pressing a button shows one green LED.

The second solution, using the test firmware, implies the computer should be able to detect the device. I don't seem to be able to find the device in Device Manager.

david-oswald, 2016-10-08:
Do you have Windows 10 anniversary update? After upgrading one of my systems to this version, I initially noticed similar problems. 

I blamed this on the fact that I had the earlier (unsigned) Chameleon driver installed (did you?). Anyway, for me this was fixed by:

- Manually installing the new driver (right-click .inf file, select install)
- Fully unplugging Chameleon and making sure it is fully powered off (removed battey)
- Re-plugged and device shows up

emsec, 2016-10-08:
Hi, we have tested with Windows 10 and there should be no problem. In the bootloader, there is no LED lit so it is a good sign that you have succeeded with the first step to trigger the bootloader. Next you have to follow the "Upgrade Procedure for Windows Users", have you done that? most importantly hav you followed Step 2:  Install the atmel_usb_dfu.inf driver (right click -> install).
After this the ATXmega128A4U DFU should appear in the device manager.  Now you can proceed with the FirmwareUpgrade.bat.

SuperVirus, 2016-10-08:
I had problems installing the driver via right click => install, too. I didn't install the libub0.dll.
Instead installed the driver from the device manager, which worked for me.

With the test-firmware windows 10 it showed up as an com port device, without installing any driver.

SuperVirus, 2016-10-08:
I had problems installing the driver via right click => install, too: I didn't install the libub0.dll.
Instead installed the driver from the device manager, which worked for me.

With the test-firmware windows 10 it showed up as an com port device "USB Serial Device (COMx)", without installing any driver, because it uses the built-in serial port class driver.

Cybersloth, 2016-10-10:
This info may help someone, Devs feel free to pull it if you don't agree

The first problem I overcame was that my browser wanted to display rather than download the .eep and .hex files, not a big problem I selected all (CNTRL A) and copied (CNTRL C), then created new files with the correct filenames and pasted (CNTRL B).  But when I ran the .bat file got the Chameleon may not be in bootloader mode message, but more importantly it mentioned line 7 in the error, this was the clue !. The Chameleon was in bootloader mode, but you can't tell as there is no feedback.  When I copied and pasted the file contents I failed to add a return (Carriage return) after the last line of both files, the .hex file would have given a line 6512 error.

The second issue was that windows did not want to install the unsigned drivers, this can be easily overcome, just google

All working very well now

Note to Devs....is there some way that the Chameleon could indicate it is in bootloader mode ?  I think some people with the issues above will assume the Chameleon has a problem as it provides no feedback

r4b3, 2016-10-10:
@SuperVirus -> when u see the com port, open install new driver and point to the directory where the drivers are.

PetersPGM, 2016-10-10:
I have installed everything according to the manual. I see neither a COMx serial port nor anything similar to ATXmega128A4U DFU in my Device Manager. I will try to find another Windows system this week to test it on that. I will also get my hands on a linux (Raspberry PI) somewhere this week too.

emsec, 2016-10-10:
@Cybersloth: the drivers ARE signed (?!)  you need the .inf and .cat in the same folder, of course. The copy paste approch is NOT recommended as it may result in corrupted firmware files (and this in turn in a defetive ChameleonMini...). Just use right mouse button and "save target as..", that works fine for me. Indication for bootloader mode is that no LED is lit AND something appears in device manager ; even without driver, a new device apperas in the device manager (for Windows). Changes are not planned: The Bootloader is factory-programmed and cannot be updated via USB.

Cybersloth, 2016-10-10:
@emsec  Unfortunately and absolutely the cut an paste will have stripped the signing, maybe a slightly more descriptive getting started would help folk like me ref the lack of indication that bootloader mode is not fed back ?

zenroth1752, 2016-10-14:
Hi,
I have seen the lsusb problem before on Win7 64 bit and Win 10 64bit cant install .inf on a 64 bit machine. I used http://zadig.akeo.ie/ from the days of HackRfOne and SDR. Download it, installed and follow the simple instructions. This will then recognise the usb device ChameleonMini and let you install the rest via Device Manager , ( Right click, browse to ChameleonMini.inf and install ) It is a simple and some say dirty trick but it works. Cheers Zen Roth

theGRK, 2016-10-15:
@zenroth1752 thanks for your suggestion, I got the same issue on win7 64bit.
When installing Zadig what driver should I better choose?

zenroth1752, 2016-10-15:
@theGRK I just installed the default WinUSB as it has the libusb1-0. 
[image]. Then you should be good to go. All the best. Zen

theGRK, 2016-10-23:
Unfortunately, I still cannot make it working on win7 x64
After installing the driver via Zadig, as @zenroth1752 kindly suggested, I see 2 different behaviours.
1. if the device is connected without holding the RBUTTON, windows identifies the device as
[image]
2.  if the device is connected holding the RBUTTON, windows identifies the device as
[image]
In any case I cannot locate the COM port to connect to and Tera Term got a grayed out menu for serial communications
[image]
Also the ChameleonFirmwareUpgrade ends with 
> There was an error with executing this command. Maybe your ChameleonMini is not in bootloader mode?

(dfu-programmer error 0xc000007b)

Any other hint?
Thanks

g-c-n, 2016-10-23:
Hi. Just in case, I was just saying in another thread that I was in a similar situation - I thought I tried everything including changing the cables. After trying a third cable out of despair, the Chameleon got detected immediately.

geo-rg, 2016-10-24:
Hi @theGRK,
it seems like you are on the right way: connecting the device holding RBUTTON is correct and the Chameleon also shows up correctly as DFU ATXMEGA128A4U. This means, that the Chameleon is in bootloader mode, where it waits for a new firmware to be flashed via USB.

Unfortunately, I do not know the dfu-programmer error. I suggest using another program for flashing instead of the .bat file, which uses dfu-programmer. An idea would be using http://www.atmel.com/tools/FLIP.aspx. In FLIP it is important that you select ("Device Selection") the correct controller (ATxmega128A4U) and then program the hex file to the flash program memory. For flashing the EEPROM ("Select EEPROM") it is important that you _uncheck_ the first two checkboxes ("Erase", "Blank check").

zenroth1752, 2016-10-25:
Hi,
Did you go into device an install the drivers? 
Go to devices the you install the chameleon drives that that will provide the bridge to the vertical port Com6. This is from memory. I will type it too properly if this does not work out.

theGRK, 2016-11-05:
Dear @geo-rg and @zenroth1752,
    sorry for the late follow-up.
After many attempts I changed the usb driver with Zadig from WinUSB to libusb-win32.
Using FLIP I programmed the chameleon, which now shows up in Windows Devices as Chameleon-Mini instead of LUFA CDC Demo.
Now I'm trying to connect with TeraTerm but the serial box is still disabled. Any suggestion here?
Thanks for your kindly help.

domoarigatosan, 2016-11-08:
@theGRK 
I has similar issue but I read out that you have reinstall the libusb-win32 from Zadig.

Try activate the ChameleonFirmwareUpgrade.bat (Doesn't matter if you can't see the serial box on TeraTerm). 

>Now your Chameleon RevG should be upgrading mode.
>Once done, close the ChameleonFirmwareUpgrade.bat
>Go to device manager > install driver > choose "have disk" > Select "ChameleonDriver.inf"
>Try communicate from TeraTerm again.

Hope it helps

geo-rg, 2016-11-21:
Sorry for the late reply. Are there any updates? Does everything work for you?

theGRK, 2016-12-04:
Sorry guys but I'm abroad without my Chameleon, I'll update you in next few weeks.

cyb4bit, 2016-12-19:
@domoarigatosan 
Very good tips i just folow the steps after the uprgrading mode: Go to device manager > install driver > choose "have disk" > Select "ChameleonDriver.inf" and now i have the Chameleon fully functional on Tera Term. I use Xp pro sp3 OS. Thanks for your apport.

geo-rg, 2017-04-05:
Since there has been no updates on this issue, I assume that everything works fine now. If not so, please re-open this issue.

closed 2017-04-05 by geo-rg

bjenky, 2021-07-01:
today i am going to fucking blow my brains out because i couldnt figure this out. this is unreasonably complicated.  i hate this fucking thing and iceman and all of you for not making this at least graspable for the common man. wtf. yall are elitist pos, and i will haunt you all. fuck the world and everything in it. i hate .all. seeya

bjenky, 2021-07-01:
FUCKING HEEEEEELLLLLPPPPPP
BJENKINSPDX@GMAIL.COM
IM TYING A FUCKING NOOSE IM FUCKING LOSING IT 
FUCK THIS COMPLICATED BS! ICEMAN FUCKING HELP ME!!!!!!!

bjenky, 2021-07-01:
I HAVE SPENT 1000$ 10000 HRS HALF A HEAD OF HAIR AND 10 SLEEPLESS NIGHTS STRAIGHT TRYING TO GET THIS OR A PROXMARK WORKING AND NOW I WANT TO END  MY LIFE
I HATE THIS SHIT
PLEASE HELP
BJENKINSPDX@GMAIL.COM