emsec/ChameleonMini #20: after setting config to ISO14443A_READER, config cannot be changed via python

cons0l3, 2016-10-08:
~>  python3 chamtool.py -v -p /dev/ttyACM0 -c ISO14443A_SNIFF
[2016-10-08 20:37:36.060585] Opening serial port /dev/ttyACM0 succeeded
[2016-10-08 20:37:36.064505] Executing <VERSION?>: 101:OK WITH TEXT
[2016-10-08 20:37:36.175404] Executing <CONFIG=?>: 101:OK WITH TEXT
[2016-10-08 20:37:36.182485] Response: NONE,MF_ULTRALIGHT,MF_CLASSIC_1K,MF_CLASSIC_1K_7B,MF_CLASSIC_4K,MF_CLASSIC_4K_7B,ISO14443A_SNIFF,ISO14443A_READER
Traceback (most recent call last):
  File "chamtool.py", line 205, in <module>
    main()
  File "chamtool.py", line 186, in main
    result = cmdFuncs[cmd](chameleon, arg)
  File "chamtool.py", line 44, in cmdConfig
    result = chameleon.cmdConfig(arg)
  File "/root/ChameleonMini.git/Software/Chameleon/Device.py", line 230, in cmdConfig
    return self.getSetCmd(self.COMMAND_CONFIG, newConfig)
  File "/root/ChameleonMini.git/Software/Chameleon/Device.py", line 177, in getSetCmd
    return self.writeCmd("{}{}{}".format(cmd, self.SET_CHAR, arg))
  File "/root/ChameleonMini.git/Software/Chameleon/Device.py", line 126, in writeCmd
    status = self.serial.readline().decode('ascii').rstrip()
**UnicodeDecodeError: 'ascii' codec can't decode byte 0x99 in position 3: ordinal not in range(128)**

working in minicom --device /dev/ttyACM0 and typing "CONFIG=NONE" I receive:
..+RNONE100:OK
or
..�.NONE100:OK

The Chamleon response with four byte values. They are change every call to CONFIG=XXX. It does not matter which config is choosen.

chamtool.py can of course skip those additional lines, but I guess it is something in the firmware...

david-oswald, 2016-10-09:
Please note that chamtool is still work-in-progress, so not expected to work without problems. Will be fixed over the coming weeks.

dev-zzo, 2016-10-12:
@cons0l3: Judging by the output you provided... Please verify you don't have log functionality enabled to log to the screen; use `LOGMODE?` to check that and `LOGMODE=OFF` to turn logging off, see the documentation for more details. Let us know if that helped.

cons0l3, 2016-10-12:
yes, LOGMODE=OFF to the rescue. No more garbeld output during chamtool.py or directly. I read in the doc that LOGMODE=OFF is the default, but do not remember ever have touched the LOGMODE setting up to this point.

see #24 I offer to change the chamtool.py to not assume LOGMODE=OFF or MEM

closed 2016-10-12 by cons0l3