emsec/ChameleonMini #21: Tutorial required

cons0l3, 2016-10-09:
A few simple examples or step by step instructions would be wonderful:
1. How to read and existing card and dump the data to a file (READER)?
1. How to take a dump and upload it to a virtual card 1-8?
1. How to clone a card from READER to directly to a virtual card?
1. How to set random UID without using buttons?
[...]

emsec, 2016-10-09:
Thanks for your constructive feedback ; however, ChameleonMini has so many use cases / things to explain that it is hard to create step-by-step instructions for every use case... the documentation is pretty complete and it took some time to write it... and it also contains a few step-by-step instructions. Getting known to Chameleon requires some "playing" I guess ; some additional examples such as uploading dumps, random UID are also in the kickstarter video (lab tests): [Video](http://www.dailymotion.com/video/x3lf45p_chameleonmini-revg-in-lab-demo-technical-explanation-of-command-line-interface_tech). The reader function as in the current firmware is still - as announced in the kickstarter campaign - quiet basic, so advanced functions (via SEND or SEND_RAW) require an understanding of NFC commands / ISO14443.  Easy functions such as IDENTIFY or DUMP_MFU to dump an Ultralight card are already provided (see here: [Chameleon Commands](https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/Page_CommandLine.html) ). For impementing tons of more reader functions we count on the open-source community: The firmware and documentation is only a starting point for everyone and you are welcome to add any new function / new card / new codec / tutorial / step-by-step infos / improved documentation etc. - we will gladly integrate it into the project.

rweijnen, 2016-10-10:
First of all, the documentation is a good starting point and has clearly taken a lot of time to write. However the documentation is written as a programmers documentation, a reference guide if you will. It explains what commands do but now which commands would be required for a certain task.

So I agree with @cons0l3 a few more examples would be very helpful and could accelerate community to start working with the chameleon and start contributing!

cons0l3, 2016-10-10:
Yes, the documentation is a good starting point. But **simple** examples for someone who just starts with NFC, ISO14443A, etc. (like me)

You would not belief the big smile I had yesterday, when I got the UID from a card. 3 other near field type cards did not work. Reasons unknown. I use my android tablet with a "nfc reader" app to dig into the cards.

To study test cases is usually a good starting point to understand the usage of a "thing". So I am considering to write a few python test cases (for learning purposes) to
a) test the firmware
b) test the python clients
So prepare and embrace yourself for incoming pull requests.

For myself, I have had a look at the firmware and was overwhelmed (for now). I will put some love into the python toolset.

gogobd, 2016-10-11:
I second the proposal of a tutorial for some ste by staep procedures. It was easy to u derstand how to clone a UID, but dumoing a whole card to emulate it is not covered well in the current documentation.

dev-zzo, 2016-10-12:
FWIW, I am putting together a tutorial here for my colleagues: https://gist.github.com/dev-zzo/88bd6e78dc62e2ed32616c4889151f8a
Let me know if I can improve it in any way. :-)

grymoire, 2016-10-17:
Suggested tip for Minicom/Linux users:

Turn on Local Echo (Control-A E) and Line Wrap  mode on (Control-A W)

johanveen, 2017-01-01:
First of all everyone the best wishes for 2017.
I must agree that the documentation for basic usage can/should be better.

There are several video tutorials but none of them explain how you can extract the dump to the computer. They only show how to make some modifications and what the possibilities are of the card.

I can make a dump_mfu of a card and upload an example dump to the chameleon mini, but I cannot download a dump to the PC (OS Ubuntu 16.04 LTS).

So if someone could help me I would be grateful.

dev-zzo, 2017-01-01:
@johanveen could you please explain in more detail the "cannot download" part? What happens when you issue the `DOWNLOAD` command?

zenroth1752, 2017-01-06:
Switch to Kali Linux - As all the NFC is set up. I have had issues with OS Ubuntu 16.04 LTS. It is a quick solution as I don't know much of the background.

johanveen, 2017-01-09:
Thanks for the suggestion. I will give it a try.

I will let you know if it was thesolution for me.

 

gr,

Johan

 

Gesendet: Freitag, 06. Januar 2017 um 23:09 Uhr

johanveen, 2017-01-09:
Goodmorning Dmitry,

 

I appologize for my late response but I was out.

I did first try a dump file from https://github.com/emsec/ChameleonMini/tree/master/Dumps.

The file I tried to upload first was MifareUltralight.mfd

 

That went oke. So I tried to make a download from a MifareUltralight card.

 

I set setting one to ISO14443A_READER

and start reading the the card with IDENTIFY

Then I did a DUMP_MFU

 

Next I used the DOWNLOAD command the get a DUMP file.

But instead of getting the card I was reading, I just get the example dump file.

 

So somehow I am not able to download the card I am trying to clone.

 

greetings,

 

Johan

 

Gesendet: Sonntag, 01. Januar 2017 um 21:37 Uhr

dev-zzo, 2017-01-09:
@johanveen I think I see the root cause behind your problem. The `DUMP_MFU` command only dumps the contents on the screen, nothing else; it does not read the contents into the onboard memory thus there is nothing to `DOWNLOAD`. You can simply save the output of `DUMP_MFU` and convert it from hex to binary for emulation later on.

manthis, 2017-02-11:
Completely agree there should be a user documentation for those like me who are novice in this domain. I think what is mostly needed is how to dump and how to restore an image.

hamsik7, 2017-12-17:
@cons0l3 did u find a way on "How to clone a card from READER to directly to a virtual card?"

hamsik7, 2017-12-17:
@dev-zzo do you have a way on how to save the output of the DUMP_MFU ?

dev-zzo, 2017-12-18:
@hamsik7 No, I am not aware of such functionality.

hamsik7, 2017-12-19:
@dev-zzo do you have any useful links or forums to help me familiarise with the use of Chameleon ?

gtpy, 2017-12-19:
@hamsik7 
Have you already checked out https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/Page_GettingStarted.html

maxieds, 2018-01-04:
### **Another Easy Method for Working with the Chameleon Mini Device ...**

**> do you have a way on how to save the output of the DUMP_MFU ?**

There is also now an open-source Android app ([ChameleonMiniLiveDebugger](https://github.com/maxieds/ChameleonMiniLiveDebugger)) [[signed apk from here](https://github.com/maxieds/ChameleonMiniLiveDebugger/blob/master/app/release/app-release.apk)] available on the [Play Store](https://play.google.com/store/apps/details?id=com.maxieds.chameleonminilivedebugger) that makes getting started with the device much easier. All it requires is the Chameleon Mini, a Droid phone, and [this usb adapter](https://www.amazon.com/gp/product/B00CXAC1ZW/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1). [#159]

hamsik7, 2018-01-05:
@gtpy  thanks for your answer ! but my problem is that i don't know how to clone a card and dump it into Chameleon :/

maxieds, 2018-01-08:
You can clone one of the preloaded card configurations with the app by clicking on the appropriate button at the top of the tools menu. Seven of these defaults are implemented in the app.

hamsik7, 2018-01-09:
@maxieds ah cool thanks ! im gonna prepare a tuto on how to clone Mifare Classic 1k and dump it in cameleon  so that those who don"t know will learn. I have another issue which is cloning Mifare DesFire ev1 any hints will help me alot 🗡 
[image]

gtpy, 2018-01-09:
@hamsik7 I have the same issue with ev1 cards. If I find a practicable solution I'll tell you. 
If you find something out it'll be great to hear from you.

hamsik7, 2018-01-09:
@gtpy yeah of course ! once i find something useful ill tell u and make a tuto about it but its been really frustrating since i searched everywhere for a practical solution but the only thing that i found in order extract a dump/clone DesFire ev1 card was a hardware solution which is basically proximark3 :/

gtpy, 2018-01-09:
@hamsik7 
where is the screenshot from?
program?
script?
[image]

Cheers

gtpy, 2018-01-10:
Ok I found it. It's from the Win GUI from bronken.
I never seen this before - i'm normally working with Linux.

hamsik7, 2018-01-10:
@gtpy can you give me a link ?

hamsik7, 2018-01-10:
@gtpy the screenshot is from GUI for using chameleon Mini on Windows.

AndreasBujok, 2018-01-14:
I see ongoing questions about an more "user friendly documentation".
Therefore i've created a friendly command list of the Chameleon Mini.

Feel free to use, addapt the current version and add and adapt new items :-)

Cheers Andreas

https://github.com/AndreasBujok/ChameleonMini

maxieds, 2018-01-15:
@hamsik7
You can now download logs and the existing card data with the [newest release (v0.1.6)](https://play.google.com/store/apps/details?id=com.maxieds.chameleonminilivedebugger) of the app. See the attached screenshot below:
[image]

CBRRider, 2018-01-25:
@dev-zzo You mentioned converting the DUMP_MFU from a hex to binary. How do you do this?

dev-zzo, 2018-01-25:
@CBRRider I use 010 Editor a lot, and there it's just Edit -> Paste from -> Paste from Hex Text. Done.

geo-rg, 2018-01-25:
@AndreasBujok Done. Thanks for it! :)

maxieds, 2018-01-26:
>You mentioned converting the DUMP_MFU from a hex to binary. How do you do
this?
If you use my android app (
https://github.com/maxieds/ChameleonMiniLiveDebugger) there's a button that
will do this in the Export tab.

CBRRider, 2018-02-09:
@maxieds So I just got my cable in but for some reason when I try and open the app on my phone it instantly closes. Any ideas? I'm using a rooted oneplus3 with oxgenos 4.5.1 with android version 7.1.1. Downloaded the app from the app store

CBRRider, 2018-02-15:
Great, looks like the app opens now. So now I was able to run the dump_mfu and dump mfu tag to bin. However, when I try to upload card  i'm getting error - invalid file size for the selected card. aborting. I switched to ultralight configuration. the file size looks to be 136 bytes. I've identified the card as a ultralight card. am i missing some step?

maxieds, 2018-02-20:
Hello.

I believe that this is a formatting issue with the Chameleon Mini Live
Debugger application. The upload / download card formats are what is
directly transferred by the board using XModem. The identify / dumpmfu
commands have a different format from this, which is to say that the dump
mfu command saves (in binary) the response given by the board.

In terms of the conversion, I will say that it could be tricky to get the
dumpmfu command to output the 8MB (or so) binary that the upload / download
card expects. Namely, the Chameleon source is written in C / C++, where
this application is on Java for Android. Not to say that this cannot be
done, but this is probably the crux of the issue you are having. Have you
looked at the NXP reader apps on play store? The formatting they give for
things like the NTAG210 chips is probably more along the lines of what the
Chameleon command is outputting. I'd be willing to include a patch to the
source once this is clarified or fixed.

Maxie

yaxax, 2019-04-01:
I still do not know how i should go further when I have key a & b in the output of the chameleon mini reve. I also have many times read the data sheet but understand a little bit of it because is in english and my english is not so good. I still don't know how to put key & a & b on a card when I get key a&b from the chameleon mini reve.Can somebody help me out whit some advice .