emsec/ChameleonMini #22: Installing the driver on Windows 7?

Xykon, 2016-10-10:
Hello,

I got the ChameleonMini today and tried it on my Notebook which is running Windows 7.

When I connect it via MicroUSB the device manager finds LUFA CDC demo with an exclamation mark. So I downloaded a zip file of the project, went to the driver folder and right-click on the ChameleonDriver.inf file. But all I get is a message that the inf file I selected does not support this method of installation.

So I downloaded the dfu-programmer package and copied dfu-programmer.exe to ChameleonMini-master\Firmware\Chameleon-Mini

I right-click on the ChameleonFirmwareUpgrade.bat and run as administrator but get this message:

"The program can't start because libusb0.dll is missing from your computer. Try reinstalling the program to fix this problem"

So I went back to "normal" mode and tried to install the driver for LUFA CDC demo but I get an error that "Windows can't verify the publisher of this driver software."

I looked through the documentation but the getting started guide doesn't seem to mention anything about how to install the drivers to get started.

r4b3, 2016-10-10:
No need for a LUFA driver.
All you need is described here
https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/Page_GettingStarted.html

Download the
- dfu Programer,
- The. bat File
- .eep and hex file
-. inf and cat file for the Chameleon

closed 2016-10-11 by david-oswald

Xykon, 2016-10-12:
I did all what is described on that page and I did manage to update. I also have the .inf and .cat files but they don't install saying "Windows can't verify the publisher of this driver software."

r4b3, 2016-10-12:
u can disable the check for driver signing
http://ccm.net/faq/3914-how-to-install-unsigned-drivers-on-windows-7
this should help

Xykon, 2016-10-12:
I tried this but it seems our IT department disabled this. Looks like I have to get another computer to try this thing.