emsec/ChameleonMini #23: Can't download XModem on OSX

gogobd, 2016-10-10:
Hi! I was trying to test the DOWNLOAD command and my cameleon waits for an XMODEM file transfer to start. I was using screen and "^a:exec !! lrx file" but it said that headers were garbled, then I tried minicom and presses ESC-Z and then R for receive and the XModem and a filename. It didn't seem to work.
Also I don't understand the documentation here very well, does that command send the whole memory that's been read from a card into a file? Can that same be done *on*card* and be stored internally? How? I'd like to clone a whole card.

gogobd, 2016-10-11:
```
110:WAITING FOR XMODEM

rx: ready to receive file
Retry 1: Got 061 sector header
Retry 0: Sector number garbled
Retry 1: Sync Error

rx: file removed.

Transfer incomplete
```

That'S what I see in screen wirh rx

david-oswald, 2016-10-11:
Not sure about this specific problem - can anyone replicate this problem? Note that there is a timeout within which the xmodem rx has to start ie you have to be relatively quick.

DOWNLOAD will receive the binary representation (an 8k slot in the FRAM) for the selected setting. This can later be uploaded (UPLOAD) to restore the card dump. For MFClassic and MFUltralight, the binary format is an mfd file as generated by other tools like libnfc.

geo-rg, 2017-04-05:
@gogobd Did you find any solution for your problem? Maybe your problem does not occur anymore with the recent firmware versions?

KubaO, 2019-03-27:
This problem seems endemic to macOS. I got it trying to download configuration from an HP 5406zl switch, using `screen /dev/tty.usbserial 115200` and then `^a : exec !! rx /users/foo/file.txt`.