emsec/ChameleonMini #24: IDENTIFY

gogobd, 2016-10-10:
For most of my cards "IDENTIFY" doesn't work, even though I see through logging that there's some "good" information coming in. That works only in reader config, right?

cons0l3, 2016-10-11:
I have a "garbled" output before and after **CONFIG=** and **IDENTIFY**
in minicom --device /dev/ttyACM0 session:

**VERSION?**
101:OK WITH TEXT
ChameleonMini RevG 161007 using LUFA 151115 compiled with AVR-GCC 4.9.2
Based on the open-source NFC tool ChameleonMini
https://github.com/emsec/ChameleonMini
commit f46f5b4

**CONFIG=ISO14443A READER**
_G@ISO14443A READER_100:OK

**IDENTIFY**
                          ▒t.▒@m▒^zH>t.@.▒v.▒101:OK WITH TEXT
MIFARE Cڱ__ 1▒  .A.▒s.A@.▒t▒6u/=$A
ATQA:   0400
UID:    504DEED2
SAK:    08

gogobd, 2016-10-11:
Do you have local logging enabled?

cons0l3, 2016-10-11:
I do not understand your question regarding local logging. I use minicom on linux and enable "local echo" by  "Ctrl-A-E". Usually I also activate "line wrap" with Ctrl-A-W. All the commands of minicom can be found in the menu "Ctrl-A-Z"

the rest is just copy&paste ;-)

the "garbled" output of the chameleon looks like uncleared output buffers **or** a read/write out of memory bounds (fencing issue), because it changes every time I execute it **or** debug output.

david-oswald, 2016-10-11:
I think what gogobd means: What is the output of "LOGMODE?" If it is not "OFF", please disable it by sending "LOGMODE=OFF".

The described behaviour is definitely not normal.

cons0l3, 2016-10-12:
Oh, LOGMODE=OFF to the rescue! It was set to LOGMODE=LIVE during my experiments (not that I remember actually setting it). That setting produced the garbled extra bytes during set UID=12345678 and CONFIG=MF_14443A_READER or IDENTIFY! 

For the chamtool.py to work properly it should 
a) issue a LOGMODE=OFF when making the connection. 
or
b) issue a LOGMODE? if it is LIVE print an error message and exit
or
c) issue a LOGMODE? if it is LIVE set to OFF or MEM

Otherwise it will choke over all "extra" bytes sent by the chameleon, as in my case! 

I can make a pull request, what should I implement a), b) or c)?

closed 2016-10-19 by david-oswald