emsec/ChameleonMini #26: Upgrade bug

Zgv163, 2016-10-11:
Hi, When i try to execute the upgrade i get the following message:

zgv@ZGV-MacBookPro:~/Desktop/upgrade$ sudo avrdude -c flip2 -p ATXMega128A4U -B 60 -P usb -U application:w:Chameleon-Mini.hex:i -U eeprom:w:Chameleon-Mini.eep:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9746 (probably x128a4u)
avrdude: NOTE: "application" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "Chameleon-Mini.hex"
avrdude: writing application (131072 bytes):

Writing | #                                                  | 2% 0.05savrdude: Error: DFU_DNLOAD failed: error sending control message: Broken pipe
avrdude: Error: Failed to set memory page 0x0000
avrdude: Error: DFU_DNLOAD failed: error sending control message: Broken pipe
avrdude: Error: Failed to set memory unit 0x00 (Flash)
 ***failed;  
avrdude: Error: DFU_DNLOAD failed: error sending control message: Broken pipe
avrdude: Error: Failed to set memory unit 0x00 (Flash)
 ***failed;  
avrdude: Error: DFU_DNLOAD failed: error sending control message: Broken pipe
avrdude: Error: Failed to set memory unit 0x00 (Flash)
 ***failed;  
avrdude: Error: DFU_DNLOAD failed: error sending control message: Broken pipe
avrdude: Error: Failed to set memory unit 0x00 (Flash)
 ***failed;  
avrdude: Error: DFU_DNLOAD failed: error sending control message: Broken pipe
avrdude: Error: Failed to set memory unit 0x00 (Flash)
 ***failed;  
avrdude: Error: DFU_DNLOAD failed: error sending control message: Broken pipe
avrdude: Error: Failed to set memory unit 0x00 (Flash)
 ***failed;  
avrdude: Error: DFU_DNLOAD failed: error sending control message: Broken pipe
avrdude: Error: Failed to write 0x0001 bytes at 0x0007
 ***failed;  
avrdude: Error: DFU_DNLOAD failed: error sending control message: Broken pipe
avrdude: Error: Failed to write 0x0001 bytes at 0x001E
 ***failed;  
avrdude: Error: DFU_DNLOAD failed: error sending control message: Broken pipe
avrdude: Error: Failed to write 0x0001 bytes at 0x003E
 ***failed;  
^C
zgv@ZGV-MacBookPro:~/Desktop/upgrade$ 

Dose any one have an idea why?

david-oswald, 2016-10-11:
This seems to be a problem specific to your installation - can you post more details like avrdude version etc.?

Zgv163, 2016-10-11:
avrdude version 6.2

And is it normal that the cameleon dosn't blink at all?

david-oswald, 2016-10-11:
Yes the device does not indicate that it is in the bootloader (both LEDs are off)

Zgv163, 2016-10-11:
I don't really understand what you mean by the device dose not indicate that it is in the bootloader?

DominikSchlecht, 2016-10-11:
Get the same (or similar) errors on Arch-Linux while trying to flash the firmware.
Command:
`sudo avrdude -c flip2 -p ATXMega128A4U -B 60 -P usb -U application:w:Chameleon-Mini.hex:i -U eeprom:w:Chameleon-Mini.eep:i`
Output:
```
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9746 (probably x128a4u)
avrdude: NOTE: "application" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "Chameleon-Mini.hex"
avrdude: writing application (131072 bytes):

Writing | #################                                  | 33% 0.68savrdude: Error: DFU_DNLOAD failed: Broken pipe
avrdude: Error: Failed to write 0x0100 bytes at 0x8700
avrdude: Error: DFU_DNLOAD failed: Broken pipe
avrdude: Error: Failed to set memory unit 0x00 (Flash)
 ***failed;
[...]
```  
avrdude-version is 6.3

tgouverneur, 2016-10-11:
I have the same issue on Linux/Gentoo with avrdude 6.3:

```
Oct 11 21:50:49 [kernel] usb 1-11: new full-speed USB device number 9 using xhci_hcd
Oct 11 21:50:49 [kernel] usb 1-11: New USB device found, idVendor=03eb, idProduct=2fde
Oct 11 21:50:49 [kernel] usb 1-11: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Oct 11 21:50:49 [kernel] usb 1-11: Product: DFU ATXMEGA128A4U
Oct 11 21:50:49 [kernel] usb 1-11: Manufacturer: ATMEL
```

Running:

$ sudo avrdude -c flip2 -p ATXMega128A4U -B 60 -P usb -U application:w:Chameleon-Mini.hex:i -U eeprom:w:Chameleon-Mini.eep:i

The only message caught by syslog is:

```
Oct 11 21:51:47 [kernel] usb 1-11: usbfs: process 4935 (avrdude) did not claim interface 0 before use
````

and avrdude output is:

```
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9746 (probably x128a4u)
avrdude: NOTE: "application" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "Chameleon-Mini.hex"
avrdude: writing application (131072 bytes):

Writing | #                                                  | 1% 0.03savrdude: Error: DFU_DNLOAD failed: Broken pipe
avrdude: Error: Failed to write 0x0100 bytes at 0x0600
avrdude: Error: DFU_DNLOAD failed: Broken pipe
avrdude: Error: Failed to write 0x0001 bytes at 0x0000
 ***failed;
avrdude: Error: DFU_DNLOAD failed: Broken pipe
avrdude: Error: Failed to set memory unit 0x00 (Flash)
 ***failed;
avrdude: Error: DFU_DNLOAD failed: Broken pipe
avrdude: Error: Failed to write 0x0001 bytes at 0x0002
 ***failed;
```

** OUTPUT STRIPPED **
```
 ***failed;  
avrdude: Error: DFU_DNLOAD failed: No such device or address
avrdude: Error: Failed to get DFU status: No such device or address
avrdude: Error: Failed to set memory unit 0x00 (Flash)
 ***failed;  
avrdude: Error: DFU_DNLOAD failed: No such device or address
avrdude: Error: Failed to get DFU status: No such device or address
avrdude: Error: Failed to set memory unit 0x00 (Flash)
 ***failed;  
Writing | ################################################## | 100% 15.05s

avrdude: 131072 bytes of application written
avrdude: verifying application memory against Chameleon-Mini.hex:
avrdude: load data application data from input file Chameleon-Mini.hex:
avrdude: input file Chameleon-Mini.hex contains 131072 bytes
avrdude: reading on-chip application data:

Reading |                                                    | 0% 0.00savrdude: Error: DFU_DNLOAD failed: No such device or address
avrdude: Error: Failed to get DFU status: No such device or address
avrdude: Error: Failed to set memory unit 0x00 (Flash)
avrdude: Error: DFU_DNLOAD failed: No such device or address
avrdude: Error: Failed to get DFU status: No such device or address
avrdude: Error: Failed to set memory unit 0x00 (Flash)
avr_read(): error reading address 0x0000
    read operation not supported for memory "application"
avrdude: failed to read all of application memory, rc=-2

avrdude done.  Thank you.
```

Zgv163, 2016-10-11:
Well I have installed avrdude using brew on my Mac and no problem occurred. We can suppose that the problem is coming from a missing driver on the Linux or a software that interrupts the process.

DominikSchlecht, 2016-10-12:
Verified, also works with Brew-installed avrdude.

Still the question is, why does not work on Linux :/

david-oswald, 2016-10-12:
Thanks for testing. We used the only minimally modified standard AVR flip bootloader, which is pretty standard. 

You can try to use dfu-programmer alternatively: https://github.com/dfu-programmer/dfu-programmer

exander77, 2016-10-12:
Same issue on Ubuntu 16.04, Avrdude 6.2.

skuep, 2016-10-12:
I noticed the "xhci_hcd" in the Kernel Log. So maybe it is an issue with USB 3.0? Can you test the Chameleon with a USB 2.0 port or USB 2.0-only computer?

skuep, 2016-10-12:
Also, what avrdude are you using (What Linux Distro+Packet Repo resp. build from sources?). According to this (old) bug, it might be a problem in a (very old, or unpatched) avrdude version.
http://savannah.nongnu.org/patch/?7896

exander77, 2016-10-12:
How to use dfu-programmer? I tried:

```
$ sudo make dfu-prog 
../LUFA/Build/lufa_build.mk:131: The XMEGA device support is currently EXPERIMENTAL (incomplete and/or non-functional), and is included for preview purposes only.
dfu-programmer atxmega128a4u erase
dfu-programmer atxmega128a4u flash-eeprom Chameleon-Mini.eep
Error while programming eeprom.
Makefile:145: recipe for target 'dfu-prog' failed
make: *** [dfu-prog] Error 1
```

david-oswald, 2016-10-14:
Yes, the above command line looks correct to me. There could be a deeper problem with DFU and USB on certain OSes. Can you try the commands directly on the command line, i.e., which output do

```dfu-programmer atxmega128a4u erase``` and ```dfu-programmer atxmega128a4u flash-eeprom Chameleon-Mini.eep```

give?

ne1, 2016-10-14:
Hi,

just tested the command under ubuntu 16.04 this is the result: 

`sudo ./dfu-programmer atxmega128a4u flash-eeprom ./../../ChameleonMini/Firmware/Chameleon-Mini/Chameleon-Mini.eep`

Checking memory from 0x0 to 0x5F...  Empty.
0%                            100%  Programming 0x60 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x800 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
Validating...  Success
0x60 bytes written into 0x800 bytes memory (4.69%).

But dmesg gives this output: 

[ 6301.853191] usb 2-2: USB disconnect, device number 32
[ 6304.523385] usb 2-2: new full-speed USB device number 33 using xhci_hcd
[ 6304.652875] usb 2-2: New USB device found, idVendor=03eb, idProduct=2fde
[ 6304.652877] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 6304.652879] usb 2-2: Product: DFU ATXMEGA128A4U
[ 6304.652880] usb 2-2: Manufacturer: ATMEL

After trying to upload firmware again with this command: sudo make program

i get this msg:

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9746 (probably x128a4u)
avrdude: NOTE: "application" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "Chameleon-Mini.hex"
avrdude: input file Chameleon-Mini.hex auto detected as Intel Hex
avrdude: writing application (131072 bytes):

Writing | ################################################## | 100% 1.55s

avrdude: 131072 bytes of application written
avrdude: verifying application memory against Chameleon-Mini.hex:
avrdude: load data application data from input file Chameleon-Mini.hex:
avrdude: input file Chameleon-Mini.hex auto detected as Intel Hex
avrdude: input file Chameleon-Mini.hex contains 131072 bytes
avrdude: reading on-chip application data:

Reading | #                                                  | 2% 0.01savrdude: Error: DFU_UPLOAD failed: error sending control message: Broken pipe
avrdude: Error: DFU status STALL
avrdude: Error: Failed to read 0x0100 bytes at 0x0A00
avrdude: Error: DFU_UPLOAD failed: error sending control message: Broken pipe
avrdude: Error: DFU status STALL
avrdude: Error: Failed to read 0x0001 bytes at 0x0003
avr_read(): error reading address 0x0003
    read operation not supported for memory "application"
avrdude: failed to read all of application memory, rc=-2

avrdude done.  Thank you.

grymoire, 2016-10-15:
I'm getting the same error while flashing on Ubuntu 16.04.1 LTS during the reading phase.
However I can connect to the device using minicom (with local echo on) and I get this

VERSION?                                                                  
101:OK WITH TEXT                                                          
ChameleonMini RevG 161007 using LUFA 151115 compiled with AVR-GCC 4.9.2
Based on the open-source NFC tool ChameleonMini
https://github.com/emsec/ChameleonMini
commit f46f5b4 

So I think I was able to flash the device.
I tried debugging the problem using the latest version of  dfu-programmer version 0.7.3 and I get
dfu-programmer  atxmega128a4u  read --debug 1
dfu-programmer: no device present.

I get the same error when trying the erase command.
I also tried the standard dfu-programmer 0.6.1 that came with Ubuntu. It can't find the device. I have USB3 ports. Does that make a difference?

david-oswald, 2016-10-16:
As Simon mentioned, USB3 could be a problem apparently - can anyone who has problems try with a USB2 port?

grymoire, 2016-10-16:
It's the USB3. I had the problem, and used a cheap USB2 hub. Upgrade worked perfectly.

skuep, 2016-10-16:
So my guess is confirmed. I Wunder whether the fault is in the OS, libusb, avrdude or the atmel bootloader. 
David, is there a known-bugs page to document the issue?

grymoire, 2016-10-16:
I successfully upgraded using the files on the GettingStarted page, but HELP does not show all of the commands the video or documentation demonstrates
For instance
help
101:OK WITH TEXT
VERSION,CONFIG,UID,READONLY,UPLOAD,DOWNLOAD,RESET,UPGRADE,MEMSIZE,UIDSIZE,RBUTTON,RBUTTON_LONG,LD
version?
101:OK WITH TEXT
ChameleonMini RevG 161007 using LUFA 151115 compiled with AVR-GCC 4.9.2
Based on the open-source NFC tool ChameleonMini
https://github.com/emsec/ChameleonMini
commit f46f5b4
GETUID
201:INVALID COMMAND USAGE
IDENTIFY
201:INVALID COMMAND USAGE

grymoire, 2016-10-16:
Dumb user error here. I did not have linewrap on. It was truncating the message.
Help says
VERSION,CONFIG,UID,READONLY,UPLOAD,DOWNLOAD,RESET,UPGRADE,MEMSIZE,UIDSIZE,RBUTTON,RBUTTON_LONG,LBUTTON,LBUTTON_LONG
,LEDGREEN,LEDRED,LOGMODE,LOGMEM,LOGDOWNLOAD,LOGSTORE,LOGCLEAR,SETTING,CLEAR,STORE,RECALL,CHARGING,HELP,RSSI,SYSTICK
,SEND_RAW,SEND,GETUID,DUMP_MFU,IDENTIFY,TIMEOUT,THRESHOLD,FIELD

grymoire, 2016-10-16:
After issuing
CONFIG=ISO14443A_READER
identify now works. Thanks for the tutorial dev-zzo !!

exander77, 2016-10-16:
I wanted to try different USB ports today, but I can't seem to be able to trigger bootloader by holding right button. What could I do?

exander77, 2016-10-16:
On USB 2.0 ports:
```
../LUFA/Build/lufa_build.mk:131: The XMEGA device support is currently EXPERIMENTAL (incomplete and/or non-functional), and is included for preview purposes only.
avrdude -p atxmega128a4u -P usb -c flip2 -U application:w:Chameleon-Mini.hex -U eeprom:w:Chameleon-Mini.eep

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9746 (probably x128a4u)
avrdude: NOTE: "application" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "Chameleon-Mini.hex"
avrdude: input file Chameleon-Mini.hex auto detected as Intel Hex
avrdude: writing application (131072 bytes):

Writing | ################################################## | 100% 1.53s

avrdude: 131072 bytes of application written
avrdude: verifying application memory against Chameleon-Mini.hex:
avrdude: load data application data from input file Chameleon-Mini.hex:
avrdude: input file Chameleon-Mini.hex auto detected as Intel Hex
avrdude: input file Chameleon-Mini.hex contains 131072 bytes
avrdude: reading on-chip application data:

Reading | ###                                                | 6% 0.05savrdude: Error: DFU_DNLOAD failed: error sending control message: Broken pipe
avrdude: Error: Failed to set memory page 0x0000
avrdude: Error: DFU_DNLOAD failed: error sending control message: Broken pipe
avrdude: Error: Failed to set memory page 0x0000
avr_read(): error reading address 0x0b9a
    read operation not supported for memory "application"
avrdude: failed to read all of application memory, rc=-2

avrdude done.  Thank you.

Makefile:134: recipe for target 'program' failed
make: *** [program] Error 1
```

On USB 3.0 ports:
```
../LUFA/Build/lufa_build.mk:131: The XMEGA device support is currently EXPERIMENTAL (incomplete and/or non-functional), and is included for preview purposes only.
avrdude -p atxmega128a4u -P usb -c flip2 -U application:w:Chameleon-Mini.hex -U eeprom:w:Chameleon-Mini.eep

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9746 (probably x128a4u)
avrdude: NOTE: "application" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "Chameleon-Mini.hex"
avrdude: input file Chameleon-Mini.hex auto detected as Intel Hex
avrdude: writing application (131072 bytes):

Writing | ################################################## | 100% 1.53s

avrdude: 131072 bytes of application written
avrdude: verifying application memory against Chameleon-Mini.hex:
avrdude: load data application data from input file Chameleon-Mini.hex:
avrdude: input file Chameleon-Mini.hex auto detected as Intel Hex
avrdude: input file Chameleon-Mini.hex contains 131072 bytes
avrdude: reading on-chip application data:

Reading | ##################                                 | 35% 0.21savrdude: Error: DFU_UPLOAD failed: error sending control message: Broken pipe
avrdude: Error: DFU status STALL
avrdude: Error: Failed to read 0x0100 bytes at 0x9200
Reading | ##################                                 | 36% 0.22savrdude: Error: DFU_UPLOAD failed: error sending control message: Broken pipe
avrdude: Error: DFU status STALL
avrdude: Error: Failed to read 0x0001 bytes at 0x0481
avr_read(): error reading address 0x0481
    read operation not supported for memory "application"
avrdude: failed to read all of application memory, rc=-2

avrdude done.  Thank you.

Makefile:134: recipe for target 'program' failed
make: *** [program] Error 1
```

exander77, 2016-10-16:
Using dfu-programer:
```
 sudo dfu-programmer atxmega128a4u erase
```
ends without any stdout without error.

```
sudo dfu-programmer atxmega128a4u flash-eeprom Chameleon-Mini.eep
```
ends with
```
Error while programming eeprom.
```
and error code 1.

Zgv163, 2016-10-17:
Well, knowing that it dose work on a Mac with Mac OSX but not on Mac with Linux it shouldn't be a problem with USB 2.0 or 3.0

tgouverneur, 2016-10-17:
Confirmed it worked with USB2

exander77, 2016-10-17:
Any other ideas to try? I am unable to flash using avrdude/dfu-programmer on both USB 2.0/3.0. I have Ubuntu 16.04 on ThinkPad T440p.

shapr, 2016-10-18:
I'm having this same issue with a ThinkPad P50 that only has USB3 ports.

exander77, 2016-10-18:
Am I the only one who can't flash even on USB 2.0 ports?

zenroth1752, 2016-10-18:
Have you tried a difference USB cable? I used Marshall gold plated should have been perfect. It was not. Tried different ones and it loaded fine. It is a simple solution. Zen

david-oswald, 2016-10-19:
If it's a problem of the USB cable then either voltage drop over the cable (very unlikely imho since the current drawn by the Chameleon is quite low) or (much more probable) some issue with the connector. MicroUSB connectors can wear off quite quickly.

In any case, the problem in this thread is not really Chameleon-specific, but should affect the combination "Flip bootloader + avrdude" in general. Maybe we should also discuss with the avrdude devs.

exander77, 2016-10-19:
Tried three different USB cables, one of them is completely new on both USB 2.0/3.0.

Errors follow:
```
../LUFA/Build/lufa_build.mk:131: The XMEGA device support is currently EXPERIMENTAL (incomplete and/or non-functional), and is included for preview purposes only.
avrdude -p atxmega128a4u -P usb -c flip2 -U application:w:Chameleon-Mini.hex -U eeprom:w:Chameleon-Mini.eep

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9746 (probably x128a4u)
avrdude: NOTE: "application" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "Chameleon-Mini.hex"
avrdude: input file Chameleon-Mini.hex auto detected as Intel Hex
avrdude: writing application (131072 bytes):

Writing | ################################################## | 100% 1.53s

avrdude: 131072 bytes of application written
avrdude: verifying application memory against Chameleon-Mini.hex:
avrdude: load data application data from input file Chameleon-Mini.hex:
avrdude: input file Chameleon-Mini.hex auto detected as Intel Hex
avrdude: input file Chameleon-Mini.hex contains 131072 bytes
avrdude: reading on-chip application data:

Reading | ################################################## | 100% 0.45s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x17f00
         0x40 != 0x00
avrdude: verification error; content mismatch

avrdude done.  Thank you.

Makefile:134: recipe for target 'program' failed
make: *** [program] Error 1
```
../LUFA/Build/lufa_build.mk:131: The XMEGA device support is currently EXPERIMENTAL (incomplete and/or non-functional), and is included for preview purposes only.
avrdude -p atxmega128a4u -P usb -c flip2 -U application:w:Chameleon-Mini.hex -U eeprom:w:Chameleon-Mini.eep

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9746 (probably x128a4u)
avrdude: NOTE: "application" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "Chameleon-Mini.hex"
avrdude: input file Chameleon-Mini.hex auto detected as Intel Hex
avrdude: writing application (131072 bytes):

Writing | ################################################## | 100% 1.53s

avrdude: 131072 bytes of application written
avrdude: verifying application memory against Chameleon-Mini.hex:
avrdude: load data application data from input file Chameleon-Mini.hex:
avrdude: input file Chameleon-Mini.hex auto detected as Intel Hex
avrdude: input file Chameleon-Mini.hex contains 131072 bytes
avrdude: reading on-chip application data:

Reading | ################################################## | 100% 0.45s

avrdude: Error: DFU_DNLOAD failed: error sending control message: Broken pipe
avrdude: Error: Failed to set memory unit 0x00 (Flash)
avrdude: Error: DFU_UPLOAD failed: error sending control message: Broken pipe
avrdude: Error: DFU status STALL
[... 188 lines cut ...]

Writing | ################################################## | 100% 1.53s

avrdude: 131072 bytes of application written
avrdude: verifying application memory against Chameleon-Mini.hex:
avrdude: load data application data from input file Chameleon-Mini.hex:
avrdude: input file Chameleon-Mini.hex auto detected as Intel Hex
avrdude: input file Chameleon-Mini.hex contains 131072 bytes
avrdude: reading on-chip application data:

Reading | ###################################                | 69% 0.26savrdude: Error: DFU_UPLOAD failed: error sending control message: Broken pipe
avrdude: Error: DFU status STALL
avrdude: Error: Failed to read 0x0100 bytes at 0x11C00
Reading | ###################################                | 70% 0.27savrdude: Error: DFU_UPLOAD failed: error sending control message: Broken pipe
avrdude: Error: DFU status STALL
avrdude: Error: Failed to read 0x0001 bytes at 0x098C
avr_read(): error reading address 0x098c
    read operation not supported for memory "application"
avrdude: failed to read all of application memory, rc=-2

avrdude done.  Thank you.

Makefile:134: recipe for target 'program' failed
make: *** [program] Error 1
```

exander77, 2016-10-19:
I am gettting error with: DFU_UPLOAD, DFU_UPLOAD or verification error. The cables works for flashing Arduinos etc.

exander77, 2016-10-21:
Did more testing, also got:
```
../LUFA/Build/lufa_build.mk:131: The XMEGA device support is currently EXPERIMENTAL (incomplete and/or non-functional), and is included for preview purposes only.
avrdude -p atxmega128a4u -P usb -c flip2 -U application:w:Chameleon-Mini.hex -U eeprom:w:Chameleon-Mini.eep

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0xffffff (probably .reduced_core_tiny) (retrying)

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0xffffff (probably .reduced_core_tiny) (retrying)

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0xffffff (probably .reduced_core_tiny)
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

Makefile:134: recipe for target 'program' failed
make: *** [program] Error 1
```

So there is surely a problem with communication. Dfu programmer does not work either. Could it be manufacturing defect? Any other ideas? I would really like to try it already.

exander77, 2016-10-24:
I was able to flash it using USB 2.0 hub.

spamv, 2016-10-26:
Thinkpad T450s
Ubuntu 14.04 -> USB 3.0 not working
Kali Virtual machine -> Working

exander77, 2016-10-26:
I did more testing, only with a single USB hub I am able to flash it, tried two other USB hubs with no lock. Actually the only working hub I have is 5 meter active USB cable.

I think this issue needs to be fixed.

skuep, 2016-10-28:
Thank you for your tenaciousness. If you now switch back to the non-working (with avrdude anyway) USB hub, can you successfully connect to the Chameleon COM Port and issue commands?

I.e. does this problem exist in combination with the bootloader firmware only and not with the chameleon firmware? If so, we can probably outrule any hardware issues.

michpappas, 2016-10-29:
Hi, same problems here. Managed to flash after trying different USB ports.

exander77, 2016-11-04:
Yes I can normally issue commands on any ports. I don't thing anybody reported that type of problem, it is clearly bootloader issue.

Also it fails randomly even with the usb hub. I think that bug is present on USB 2.0/3.0 with hub/without hub. On some combinations I have the bug all the time, on some I have it rarely.

Vge0rge, 2016-11-12:
Same problem here, my specs are:
Lenovo T450 
Fedora 24
Usb 3 port 
avrdude: Version 6.1

I tried to flash the firmware with avrdude, the first time I didn't realize that I didn't have the avra package installed. Although I got the same errors about the broken pipe as all other people, I reconnected the chameleon and its was recognized by dmesg  as Kapser And Oswald device. I successfully connected with it through minicom and started to type commands, the only response  I have gotten was 101:OK. 

Following that I realized that I didn't have avra installed, so I installed it and retried. The same errors occurred during the firmware flashing. Now I am getting an "device descriptor read/64, error -110"    at my dmesg which  is most probably a power relating error and I hope that it is a result of invalid flashing. 

Anyway, I just wanted to state these. I don't have a usb2 port to try as the other stated but I will try flashing through a virtual machine  or windows and I will tell here what it will happen.

gtpy, 2017-11-05:
Same issue with Kali rolling.
No chance to flash it under Linux.
3.0, 2.0ports.
Different cables.
Under windows same Portale cable flash complete.

geo-rg, 2017-11-06:
@gtpy As you noticed, this seems to be a problem with linux and its usb support since it works with the same hardware but with different OS. Thus, we cannot do here anything.

gtpy, 2017-11-06:
@geo-rg thanks. So I try it with Windows in virtual box.

gtpy, 2017-11-08:
.I solved this problem for:  ubuntu 17.10 + kali rolling -->  `sudo apt install libusb-dev`

`avrdude -c flip2 -p ATXMega128A4U -B 60 -P usb -U application:w:Chameleon-Mini.hex:i -U eeprom:w:Chameleon-Mini.eep:i
`

If it doesnt work well, I changed bitclock to different values(-B 60, -B 64, -B 60) and it works well.
Maybe it is in some instances too fast or too slow?

geo-rg, 2017-11-09:
@gtpy Thanks for your hint! 

Since I read [this](http://www.openwall.com/lists/oss-security/2017/11/06/8) a few days ago, I'm nearly sure that the linux usb subsystem causes these errors.

skuep, 2017-11-10:
It's interesting that changing the -B parameter changes behavior at all. It is supposed to set the click speed when using the hardware isp interface. Someone should check what the parameter does when using -c flip2. 

As easy as it sounds, I currently would not target the linux USB stack as the culprit. 
Best regards

gtpy, 2017-11-10:
Maybe it was a coincidence that it works after different tries. I'll try to go to an detailed analysis if I find some more time - are actual really busy in Job and family

Lipovlan, 2024-07-05:
I am also experiencing the same problems with both 3.0 and 2.0 USB A ports on my Lenovo E580 with Debian 12.5 and 6.1.0-21-amd64 kernel.

Using a USB 2.0 Hub in the USB 3.0 port has helped solve this issue for me.