emsec/ChameleonMini #27: Reading data from MIFARE Classic 4K

comeradealexi, 2016-10-11:
Been playing around with my Chameleon and it's really cool, it's identified all of my NFC cards and it's pretty interesting! 

I can't however figure out how to read the contents of the card or how to clone it. I understand the DUMP_MFU is only for the Lightweight cards so with the only cards I've got being MIFARE Classic 4K and MIFARE DESFire EV1 will I have to wait for new code to read from these cards?

Hope this isn't a silly question, I'm new to this NFC stuff :)

dev-zzo, 2016-10-12:
The problem here is that before reading/writing to MIFARE cards except Ultralight, you have to authenticate to the card with "secret" keys. No keys = no reading. The only way would be to implement the whole authentication part (reader side) in firmware and then somehow load keys, and only then you'll be able to read/write/clone cards. Bummer, eh?

david-oswald, 2016-10-12:
If you have a separate reader, libnfc can be used to dump MFClassic (https://github.com/nfc-tools/libnfc). 

Btw, the reader side would only help to dump, to get the keys one would need to use the well-known key extraction techniques for MFClassic.

closed 2016-10-12 by david-oswald