emsec/ChameleonMini #30: ChameleonMini GUI Tool for easier operation

bronken, 2016-10-12:
If anyone has use for this:
bronken.de/chameleonminigui

Its not finished but since SW0002 it contains almost all commands currently available. Up/Downloading of Files are already on my "to do" list.

I didnt know where to publish. Its no a real issue but not a pull request either, but should be somewhere around this github project because of its nature.
If this is not wanted here, any admin can feel free to delete this post.

dev-zzo, 2016-10-13:
Why not make a proper github repo for this, with source and binary releases?

david-oswald, 2016-10-14:
I agree, could be a Github repo, which we then can link in the Wiki / on the homepage. For now, I've linked your page.

closed 2016-10-14 by david-oswald

gtpy, 2018-01-10:
maybe really old issue and not sure if user bronken is active at github but @bronken can you give out the source code so we can add some functions or use it as a base for new.
Alternatively we need to start from scratch.
Thanks.

maxieds, 2018-01-10:
This is basically what I've already been doing recently except on an [Android phone platform](https://github.com/maxieds/ChameleonMiniLiveDebugger) [play store screenshots](https://play.google.com/store/apps/details?id=com.maxieds.chameleonminilivedebugger). The source is in Java, so why don't you just use my source code base once it's a little more stable and restyle the GUI toolkit from android to something more desktop'y? You can actually google around and find links to running android apps on the Linux desktop, so it can't be that far off. Actually, I wouldn't mind getting involved in the development of this project.

gtpy, 2018-01-10:
Yeah you are right. I’ll work on it. Java isn’t my best language but I think we can build on it.

ikarus23, 2018-02-08:
What's about https://github.com/iceman1001/ChameleonMini-rebootedGUI (Windows only)? Has anyone tested it (I have no ChameleonMini)?

gtpy, 2018-02-08:
@ikarus23  I take a quick look at it - but actually (I'm not sure why) it is not working correctly with the original RevG only with RevE rebooted.

We can take some inspiration of it and build a nearly identical GUI for RevG if needed?

I like their ideas and would go for it to build a GUI which could be a combination of the bronken product and the RevE rebooted GUI, but only for usage RevG

ikarus23, 2019-02-01:
@gtpy Just FYI: There has been a new release at https://github.com/iceman1001/ChameleonMini-rebootedGUI/releases. It claims it does support the RevG firmware now.

gtpy, 2019-02-01:
Yeah, thanks I’ve already seen this.
My GUI is available under gitlab.
https://gitlab.com/Gtpy/ChameleonMini-RevG_GUI

Use the one you like best. I like and use both :)

ikarus23, 2019-02-01:
Oh, great! I didn't know you implementation of a ChameleonMini GUI. Thank you for your work.