emsec/ChameleonMini #31: Support for MIFARE DESFire

dev-zzo, 2016-10-13:
I would very much like to see Chameleon supporting emulation of DESFire cards. I reckon this was already implemented according to the WISSEC paper, but is not shred with general public. Any plans on putting it on github?

exander77, 2016-10-13:
I second this. MIFARE DESFire is a must.

andruszd, 2016-10-13:
Me to for MIFARE DESFire EV1 and MIFARE DESFire EV2  versions

david-oswald, 2016-10-14:
I agree - however DESFire has a very complex protocol (three different modes of communication to start with, many functions, ...) -> we will likely start with a subset (in the WISSEC paper, it was mainly just authentication). If anyone wants to further contribute here pls contact us (or do it and send us a pull request for the complete DESFire emulation :) )

Hoopy-Frood, 2016-10-14:
Thanks David!  Authentication is a good starting point.  I will be sending a pull request for this.  I think as long as we have authentication, we can used a hardcoded file system to emulate a target cards.

Hoopy-Frood, 2016-10-14:
Thanks David!  If I can have the authentication code, I think I can hard code the file system of a target card.  Please do see if you can include the DESFire authentication in the code.

dev-zzo, 2016-10-14:
FYI, this is what's lying around the 'nets:
http://www.ti.com.cn/cn/lit/an/sloa213/sloa213.pdf
Quite nice for TI guys to provide everyone with such a nice reference implementation... ;-)

dev-zzo, 2016-10-14:
Oh well. I'll get things rolling in my fork.

dev-zzo, 2016-10-14:
By the way, any suggestions for a good (= compliant, stable, documented APIs) reader for these cards? Will have to test with something, right? AFAIK, proxmark3 does not support DESFire cards, and that's the only one "reader" I have.

david-oswald, 2016-10-14:
libfreefare has DESFire support, so it works with any libnfc-compliant reader. The libfreefare code could also be useful as a reference how the reader side works.
However there are some quirks in the freefare code, e.g., it forces the DESFire to speak at 424 kBit/s, which is not supported on the Chameleon currently. So you'd have to patch the respective code parts to use 106 kBit/s. libfreefare uses the APDU-wrapped mode by default, ie puts the native DESFire commands into an APDU (Note: there is also an APDU mode with ISO7816 commands, I think this is used rather infrequently in reality though). I think a DESFire emulation should support at least the native commands and the APDU-wrapped ones. So there probably have to be RX/TX functions that handle this transparently.

dev-zzo, 2016-10-18:
Since everyone is really looking forward to having support for DESFire in Chameleon, I think it is OK to ask for some help. :-) Could anyone who has the actual cards do a few captures of PDU traffic between cards and readers, for reference? Selecting the card, authenticating, selecting an app, reading a file... that kind of stuff. Including the 14443-4 headers please. Bonus points for full captures of 14443-4 and 7816 exchanges! Thanks!

FWIW, I think I have implemented ISO 14443-4 layer and stuff below it, so I'm working on implementing DESFire commands now. Not sure at which point it will be ready for testing, but I am confident we'll have something before mid-November.

david-oswald, 2016-10-19:
Maybe the captures here help:

http://adamsblog.aperturelabs.com/2013/02/atmel-sam7xc-crypto-co-processor-key.html

Otherwise give us a few days and we'll get/find something.

martinbeier, 2016-10-19:
@dev-zzo thanks for your effort on the DESFire stuff.
afaik pm3 has some DESFire support...

If you still need captures of DESFire card communication, I could contribute.
I have some DESFire + DESFire EV1 + DESFire EV2 cards.
I think it would be handy to have at least the common command set, that all DESFire cards support.
(native commands and the APDU-wrapped)

dev-zzo, 2016-10-19:
@david-oswald Thanks for the link, I will go through it ASAP!

@martinbeier Would be awesome if you could play with them cards and share PDU traces. As for the command set, I am currently at the basic DESFire level; I believe EV1 and EV2 are supersets but I don't have docs to back that.

martinbeier, 2016-10-19:
@dev-zzo I can confirm your assumption, DESFire EV1 and EV2 are fully backward compatible

dev-zzo, 2016-10-20:
And we have the first contact, ladies and gentlemen.

```proxmark3> hf 14a reader
 UID : de ad ba be 11 22 33
ATQA : 03 44
 SAK : 20 [1]
TYPE : NXP MIFARE DESFire 4k | DESFire EV1 2k/4k/8k | Plus 2k/4k SL3 | JCOP 31/41
MANUFACTURER : no tag-info available
 ATS : 06 75 00 81 02 00 6e 79
       -  TL : length is 6 bytes
       -  T0 : TA1 is present, TB1 is present, TC1 is present, FSCI is 5 (FSC = 64)
       - TA1 : different divisors are supported, DR: [], DS: []
       - TB1 : SFGI = 1 (SFGT = 8192/fc), FWI = 8 (FWT = 1048576/fc)
       - TC1 : NAD is NOT supported, CID is supported
       -  HB : 00
Answers to chinese magic backdoor commands: NO```

And the card's view:
```ISO14443-3: In IDLE, RX'd 52
ISO14443-3: In READY1, RX'd 93
ISO14443-3: In READY1, RX'd 93
ISO14443-3: In READY2, RX'd 95
ISO14443-3: In READY2, RX'd 95
ISO14443-3: In ACTIVE, RX'd E0
ISO14443-4: In EXPECT_RATS, RX'd E0
ISO14443-3: In ACTIVE, RX'd 40
ISO14443-4: length fail
ISO14443-3: In ACTIVE, RX'd 43
ISO14443-4: length fail
ISO14443-3: In ACTIVE, RX'd 50
ISO14443-3: Got HALT```

Now to find a reader that would issue actual commands. :)

skuep, 2016-10-20:
Quite impressive! Like to see the way you implemented it in your Code.

dev-zzo, 2016-10-20:
@skuep The code is public at https://github.com/dev-zzo/ChameleonMini/tree/desfire

Maxhy, 2016-10-20:
That looks great :smile:.
I will gladly give you some support in two weeks @dev-zzo once I receive the ChameleonMini (I know well the protocol as I implemented all DESFire EV1 communication into [LibLogicalAccess](https://github.com/islog/liblogicalaccess) library, hope I can be from some help) and free time. If you haven't finished it yet so you're fast haha.

dev-zzo, 2016-10-21:
@Maxhy I will give LibLogicalAccess a close look this weekend. :-)

geo-rg, 2016-10-24:
This is very impressive, we really like your work and look forward to any progress you make!

dev-zzo, 2016-10-24:
The current implementation will be able to support emulation of EV0 cards completely, and EV1 cards to some extent. Currently, I don't have insight into how key material is managed for EV1/EV2 cards, specifically the nuances of setting the key type... but I will find out eventually. >:-)

For EV0, the remaining parts are the actual FS interactions (values/record files, read/write operations, transactions). Authentication, key material management, and application management should be ready for testing; if anyone is willing to test and debug this please do so.

exander77, 2016-10-24:
Can't compile your repo, I am getting:
```
avr-gcc -c -pipe -gdwarf-2 -g2 -mmcu=atxmega128a4u -fshort-enums -fno-inline-small-functions -fpack-struct -Wall -fno-strict-aliasing -funsigned-char -funsigned-bitfields -ffunction-sections -I. -I../LUFA/.. -DARCH=ARCH_XMEGA -DBOARD=BOARD_NONE -DF_USB=48000000UL -DF_CPU=27120000UL -mrelax -fno-jump-tables -x c -Os -std=gnu99 -Wstrict-prototypes -Wall -DUSE_LUFA_CONFIG_HEADER -DFLASH_DATA_ADDR=0x10000  -DFLASH_DATA_SIZE=0x10000  -DSPM_HELPER_ADDR=0x21FE0  -DBUILD_DATE=161025 -DCOMMIT_ID=\"4032cd8\" -DCONFIG_MF_CLASSIC_1K_SUPPORT -DCONFIG_MF_CLASSIC_1K_7B_SUPPORT -DCONFIG_MF_CLASSIC_4K_SUPPORT -DCONFIG_MF_CLASSIC_4K_7B_SUPPORT -DCONFIG_MF_ULTRALIGHT_SUPPORT -DCONFIG_MF_DESFIRE_SUPPORT -DCONFIG_ISO14443A_SNIFF_SUPPORT -DCONFIG_ISO14443A_READER_SUPPORT -DSUPPORT_MF_CLASSIC_MAGIC_MODE -DSUPPORT_FIRMWARE_UPGRADE -DDEFAULT_CONFIGURATION=CONFIG_NONE -DDEFAULT_RBUTTON_ACTION=BUTTON_ACTION_CYCLE_SETTINGS -DDEFAULT_LBUTTON_ACTION=BUTTON_ACTION_RECALL_MEM -DBUTTON_SETTING_GLOBAL -DDEFAULT_RED_LED_ACTION=LED_SETTING_CHANGE -DDEFAULT_GREEN_LED_ACTION=LED_POWERED -DLED_SETTING_GLOBAL -DDEFAULT_LOG_MODE=LOG_MODE_OFF -DLOG_SETTING_GLOBAL -DDEFAULT_SETTING=SETTINGS_FIRST -DDEFAULT_PENDING_TASK_TIMEOUT=10  -DENABLE_EEPROM_SETTINGS  -MMD -MP -MF Bin/Commands.d Terminal/Commands.c -o Bin/Commands.o
Terminal/Commands.c: In function ‘CommandGetVersion’:
Terminal/Commands.c:24:3: error: invalid initializer
   snprintf_P(OutParam, TERMINAL_BUFFER_SIZE, PSTR(
   ^
../LUFA/Build/lufa_build.mk:294: recipe for target 'Bin/Commands.o' failed
make: *** [Bin/Commands.o] Error 1
```

geo-rg, 2016-10-25:
@dev-zzo It looks like you have your old Makefile fix for compiling under Windows, which breaks up compiling under linux.

dev-zzo, 2016-10-25:
@geo-rg @exander77 I have rebased my branch to master's HEAD, it should be OK now.

dev-zzo, 2016-10-25:
I think it's best if you create issues in my fork for any problems you might discover. :-)

exander77, 2016-10-25:
I don't see issue section on your repository. But it compiles fine now. I will try DESFIRE emulation and will report back.

exander77, 2016-10-25:
I am looking at ATS values, the emulation gives me 7500810200, normal Desfire cards gives me 7577810280, same as here: http://nfc-tools.org/index.php?title=ISO14443A

doegox, 2016-10-26:
second byte is the "speed" byte and as the current emulation only supports the slowest speed, @dev-zzo prefers to indicate to the reader to not try higher speeds. Last byte changes too as it's a checksum.

exander77, 2016-10-26:
I just tested emulation on some doors, I had success with some doors, but not with all, maybe related to the speed?

dev-zzo, 2016-10-26:
@exander77 From what I know, the 2nd byte in ATS reports which comm speeds the card supports, as correctly noted by @doegox ; as Chameleon right now doesn't support anything faster than the basic 106k/s, that's what is hardcoded in my emulation. If the reader tries to use higher speed then it won't work anyway, so there is no big point in reporting faster speeds. The last byte reportedly is unused and can be set to anything, but I will set ti to 80 to better mimic the original hardware.

dev-zzo, 2016-10-26:
@exander77 please capture the exchange for the ones that didn't work; you can use proxmark3 for that.

dev-zzo, 2016-10-28:
For those trying things and seeing bugs: use the built-in logging facility to save traffic for future analysis. Before presenting the card, use `LOGMODE=MEMORY` to save the data, then use `LOGDOWNLOAD` to get it from the card to your box. Use `chamlog` to read the trace. This can provide the insight needed to fix things! Thanks :-)

dev-zzo, 2016-10-31:
Alright, we're battling with auth code now... I am not exactly sure what the problem is, though. If I understand the D40 algorithm correctly, then on the first step, the card sends E(RndB) without using CBC at all (the IV is not updated). On step 2, the reader sends E(RndA|RndB') using CBC with zero IV -- and this is the only mention of CBC in the whole chapter on auth in the doc. On step 3, the card sends E(RndA'), again without using CBC. This does align with the example here: https://ridrix.wordpress.com/2009/09/19/mifare-desfire-communication-example/ -- if you do each step manually data agrees well (assuming the key of all zeros).

The problem is, libfreefare doesn't agree with this flow: https://github.com/nfc-tools/libfreefare/blob/master/libfreefare/mifare_desfire.c#L366 seems to use CBC all the way through the three steps.

I have come up with a method to test whether at least parts of crypto work as expected... We can easily figure out what RndB is since we know the key the card uses and can decipher the value from the data being sent over the wire. We can also figure out RndB' from what the reader sends; since CBC is used in send mode, we can use the first block E(RndA) as the IV and get the plaintext for 2nd block easily as well. So we can verify RndB being handled correctly at all times if we know the key; additionally, this does not depend on whether CBC is used throughout or only on step 2. This seems to work on the data in Ridrix's blog page but not on e.g. Adam's blog and not in tests with a reader we have. Why? No idea whatsoever. Any insights from you guys?

doegox, 2016-10-31:
All steps are with CBC, ridrix [told it in a comment](https://ridrix.wordpress.com/2009/09/19/mifare-desfire-communication-example/#comment-27)
Beware older and newer Desfire don't use the same auth protocol: older are using the trick described in ridrix comment and newer are using a regular CBC as you may expect. cf https://github.com/nfc-tools/libfreefare/blob/master/libfreefare/mifare_desfire.c#L406

doegox, 2016-10-31:
Also beware when testing 3DES with a null key, it has the nasty property that ENC(data)==DEC(data).
So maybe a source of confusion is also when you assume an encryption instead of a decryption at some step -> it will work on Ridrix data but it's wrong with any non null key.

dev-zzo, 2016-10-31:
@doegox Thanks for the response. Do you know more about "older and newer" cards? How old/how new? Is this DESFire EV0 vs EV1 or is this within the same major version?

dev-zzo, 2016-10-31:
I have made a test script using pycrypto: https://gist.github.com/dev-zzo/87e0947f3ca0bb6d6baf78dd4d0ecb9c
It demo's everything I don't understand about the algorithm. :-)
* If CBC was always used, as libfreefare code suggests, then Adam's data would decode correctly in the right column throughout, which is not the case.
* If CBC were used partially, as the D40 docs suggest, then data in the left column would be correct except the third line, which is the case for Ridrix data, but again not the case for Adam's.
* There is also an option of Adam's data being intentionally broken.

doegox, 2016-10-31:
afaik libfreefare is working fine, so better to take it as a model ;)
sorry, about to leave for holidays, can't help you right now.

I was not correct about older & newer cards, it's not about cards but commands.
AUTHENTICATE_LEGACY == 0x0A is using this weird "reverse" CBC mode
AUTHENTICATE_ISO == 0x1A and AUTHENTICATE_AES == 0xAA use proper CBC mode

dev-zzo, 2016-10-31:
@doegox I am currently trying to get the "legacy" mode supported properly, as it is the only one supported by EV0 cards. ISO and AES auth modes are supported from EV1 onwards AFAICT.
I guess I need more input data here...
Have fun on your holidays!

dev-zzo, 2016-10-31:
I'll just leave this here...
http://en.pudn.com/downloads134/ebook/detail572228_en.html

doegox, 2016-10-31:
A quick test with my own key tracing a DF EV1
```
do_desfire("00112233445566770011223344556677", "079163698cc81654", "fdd62db6b21f2d99", "268577f5d11b1e13", "487dae253a1455ca")
```
This matches the behavior of ridrix data.
I tried several variations on Adam's data but could not make sense of it neither.

dev-zzo, 2016-11-01:
@doegox Then it seems like libfreefare has an issue handling the legacy authentication method?

dev-zzo, 2016-11-02:
Just to let everyone here know, EV0 "legacy" authentication has been verified to work fine. My big thanks to @shombre for help with testing it!

arne182, 2016-12-01:
Hello @dev-zzo I wanted to reproduce your https://gist.github.com/dev-zzo/77e1365d40b0a70f8d33e988412afcb3/revisions proxmark3 read. Is this still working with your Latest commit b73bba3?

dev-zzo, 2016-12-01:
@arne182 The commit b73bba3 completely breaks *only* file-related functionality, which is not finished yet anyway. :-) The basic functionality of getting the PICC identified should work -- please open a bug at my fork if it does not.

arne182, 2016-12-01:
I tried it with another reader with the pn532_uart it is responding but it fails to respond on the samsung galaxy s4. The s4 should be able to read the uid but the tag is not found. Using the same UID on the MF_CLASSIC_1K_7B config, android reports the UID correctly.

Reventl0v, 2018-05-01:
Hi, is it possible right now to "emulate" a desfire ev1 card (only the part related to the UID) ? Simply answering when asked for the (U)ID (with ATQA 0x0344 and SAK 0X20) ?

R