emsec/ChameleonMini #32: MF_CLASSIC_4K data download / upload

Xykon, 2016-10-14:
Hello,

I've finally gotten the ChameleonMini to work under Linux / VMWare. I'm now trying to emulate a Mifare 4K card but I'm having trouble uploading the data.

I flashed the Chameleon with the latest github firmware, created a card in slot 0 with CONFIG=MF_CLASSIC_4K and gave it a UID.

When I download the memory using minicom, I get a file which is 8k in size, but the dump I created of the original card using mfoc -O card.mfoc is only 4k. When I upload these 4k via xmodem then the resulting mfoc -O clone.mfoc is not identical with the original dump, it seems the data after 0x400 is missing.

Why does the download command receive 8k of data for a 4k card? And how do I need to modify the 4k dump from mfoc to upload into the Chameleon memory?

david-oswald, 2016-10-14:
Every slot in the FRAM is 8k - this is why DOWNLOAD gives 8k back. The first 4k should be the card, rest zero in this case.
The missing data after 0x400 sounds like a bug to me, we'll look into this.

Xykon, 2016-10-15:
Just a quick update... I had time today to do the same test with a 1k mifare card.

Read the card with mfoc -O card1k.mfoc, changed CONFIG=MF_CLASSIC_1K and uploaded the mfoc file.

When I then read the Chameleon with mfoc -O clone1k.mfoc it's identical to the original card1k.mfoc

So this seems to be a bug specifically with the MF_CLASSIC_4K emulation.

david-oswald, 2016-10-16:
OK, Georg will take care of this after his return from holiday. Can you supply an example file? (replace the keys / UID with FF ... FF) if they are confidential)

geo-rg, 2016-10-18:
I'm sorry, but I cannot reproduce your error. The first thing I tried was:
1. CONFIG=MF_CLASSIC_4K
2. UPLOAD
3. upload sample MFC 4k card
4. read out dump with nfc-mfclassic
With this, the uploaded dump and the read dump are exactly the same, including non-zero bytes after 0x400. Then I tried the same but dumped with mfoc. Again, both dumps are the same.

As David suggested, could you provide an example file (with replaced keys/UID if necessary)?

Xykon, 2016-10-24:
Sorry it took some time to test this again but work kept me busy as usual.

This seems to have been a problem with mfoc rather than the Chameleon. When I initially tried this with the latest mfoc from github I got an error message "Error, cannot write dump" so I used an earlier release instead.
However I've tried the latest release again today and even though it still gives the "Error, cannot write dump" message it does actually write the dump and correctly this time.

Both 1k and 4k cards verify against the original cards.

Thanks a lot for testing this on your end as well.

closed 2016-10-25 by geo-rg