emsec/ChameleonMini #34: Logging: too tailored to MIFARE Classic?

dev-zzo, 2016-10-14:
Looking at possible values for log entries, I see those application related are purely for MIFARE Classic. Would it be possible to restructure the log entry codes so that applications could define their app-specific codes instead?

I am a bit confused as to what would be the best approach to use logging in the DESFire application I am writing.

david-oswald, 2016-10-14:
The application-related log entries are necessarily specific to a card (ie MFClassic or Ultralight...). Just add DESFire entries there.

skuep, 2016-10-14:
I would suggest implementing logging into your application at a very late stage of development. 
However, most of the application log entry types are not specific to mifare, and it was never the intention. Just add whatever you feel necessary for DESFIRE. If you feel it is really too DESFIRE specific, you could name the entry LOG_INFO_APP_DESFIRE_SOMETHING.
Technically there should be space for 64 application log entry types for type LOG_INFO_* in total in the current implementation and another 64 app entry types for LOG_ERR_*.

closed 2016-10-16 by david-oswald