emsec/ChameleonMini #41: Macbook connection issue

willpowell8, 2016-10-17:
I connect by board in DFU mode (no lights on to my mac) having installed avrdude with brew and get the following:
`MacBook-Pro-3:Chameleon-Mini Keytree20$ sudo avrdude -c flip2 -p ATXMega128A4U -B 60 -P usb -U application:w:Chameleon-Mini.hex:i -U eeprom:w:Chameleon-Mini.eep:i

avrdude: Error: No matching USB device found
avrdude: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude: error reading signature data for part "ATxmega128A4U", rc=-1
avrdude: error reading signature data, rc=-1

avrdude done.  Thank you.`

Not Sure how to resolve.

david-oswald, 2016-10-17:
Can you check if any of the solutions in https://github.com/emsec/ChameleonMini/issues/26 applies?

willpowell8, 2016-10-17:
It looks like my issue is that it device isnt recognised on USB. Not sure why... any ideas?

david-oswald, 2016-10-19:
What does dmesg output when you plug in the device? What about lsusb?

hydrotik, 2016-10-19:
Just got mine today and having same issue. Excited to get this up and running! :)

zetteeh, 2016-10-20:
the macbook i'm using doesn't even have usb3.0.. the error is exactly the same. (osx 10.8, avrdude version 6.2)

zetteeh, 2016-10-20:
worked with a fresh new cable!

pktmterenceg, 2016-10-21:
+1 for wilpowell8's issue on Mac. (Mac OS X 10.11.6).

Cable seems OK otherwise. Will try another if that's the general consensus. 

Thanks!

g-c-n, 2016-10-23:
Just to add my 2 cents, I was in the same situation and had tried with 2 different cables, After an hour of scratching my head I decided to try a third cable and... It worked perfectly fine. Who knew these cables could be so dodgy?

pktmterenceg, 2016-10-24:
Ok, since we're narrowing in on a possible solution here, would you care to endorse the brand/model of cable that worked for you? Please & thank you.

fergus, 2016-10-24:
I had the same issue. It worked for me once I unplugged and replugged the reader.

willpowell8, 2016-11-01:
I have changed the cable, replugged still with no joy.

hydrotik, 2016-11-22:
I've tried 4 cables now and nothing...

geo-rg, 2016-11-28:
@g-c-n What cable did you use?

pktmterenceg, 2017-11-05:
Bump. I never solved this, and then packed everything up for a big move. Just now getting back to it, and even with new cables, I'm still exactly where I was a year ago. I'd really, really like to get this working.

Thanks.

WolfgangMau, 2018-01-12:
same problem on OSX (El Captain)
/dev/tty.usbmodem1421 is present if the chameleon is plugged in 
the cable works perfect with my pm3

hydrotik, 2018-01-12:
@david-oswald Any input on this? Seems like a number of folks are having this issue and this thing is sitting in my desk wasting away. Maybe @zetteeh you can provide some info on how you got this up and running for the rest of us?

WolfgangMau, 2018-01-12:
ok, looks like a 'layer 8 problem' ( Human error )
I have a 'ChameleonMini RevE Rebooted'  - I think this device behaves different from the one which is covered here.
my fault

AndreasBujok, 2018-01-26:
I'm now on the same situation :-(
Don't work on MAC OS SIERRA 10.12.6.

Any solutions?

```
sudo dmesg | grep -i cham
stateNumber is On012123.201667 Chameleon-Mini@fd131000: AppleUSBDevice::ResetDevice: <software attempt to RESET>
012123.221647 Chameleon-Mini@fd131000: AppleUSBDevice::ResetDevice: <software attempt to RESET>
012123.241634 Chameleon-Mini@fd131000: AppleUSBDevice::ResetDevice: <software attempt to RESET>
012123.261752 Chameleon-Mini@fd131000: AppleUSBDevice::ResetDevice: <software attempt to RESET>
012123.281677 Chameleon-Mini@fd131000: AppleUSBDevice::ResetDevice: <software attempt to RESET>
012123.301712 Chameleon-Mini@fd131000: AppleUSBDevice::ResetDevice: <software attempt to RESET>
012123.453984 Chameleon-Mini@fd131000: AppleUSBDevice::ResetDevice: <software attempt to RESET>
012123.473551 Chameleon-Mini@fd131000: AppleUSBDevice::ResetDevice: <software attempt to RESET>
012123.493673 Chameleon-Mini@fd131000: AppleUSBDevice::ResetDevice: <software attempt to RESET>
012123.513620 Chameleon-Mini@fd131000: AppleUSBDevice::ResetDevice: <software attempt to RESET>
012123.533642 Chameleon-Mini@fd131000: AppleUSBDevice::ResetDevice: <software attempt to RESET>
012123.553615 Chameleon-Mini@fd131000: AppleUSBDevice::ResetDevice: <software attempt to RESET>
```

AndreasBujok, 2018-01-26:
BTW, 

the funny thing is, that the upgrade process with avrdude works very well without any issues :-O

AndreasBujok, 2018-02-01:
Very strange.

After installing the driver for the PL2303 (https://www.adafruit.com/product/954) i got back the device "usbmodemFD1311" for my ChameleonMini :-)

WTF?!?

maxieds, 2018-02-01:
> product/954) i got back the device "usbmodemFD1311" for my ChameleonMini

When I had the stock Atmel drivers for the Chameleon installed on a Windows
10 box, installing the Google adb USB drivers for Android Studio wiped out
my device. We're you using it on the same USB port as something else?

AndreasBujok, 2018-02-02:
Hi @maxieds.

Sorry, but i don't understand your question?!?
I'm using a MacBook with SIERRA 10.12.6.

Cheers Andreas

hamsik7, 2018-02-02:
Hey guys why won't you just install a Virtual Machine (Linux/Win) on ur Macs.
Linux if you want to use commands and for Win there is GUI so its up to u to choose !

pktmterenceg, 2018-02-02:
While we Mac users could do that, I don’t think that’s the right answer. Here’s why:

1. There is some underlying issue keeping (would-be) host OS from seeing the device, and this issue really ought to be properly understood and addressed. 

2. If the host can’t see a USB device, what causes you to believe that a VM - which inherently relies upon the host for basic USB services - would be able to see the same device?

banannd, 2019-03-21:
I had the same problem in Ubuntu 16.4 "avrdude: Error: No matching USB device found"
1. I changed the cable but it did not help,
2. then I connected to the device "socat - / dev / ttyACM0, crnl" and once again I entered the command "upgrade"
then everything was ok.