emsec/ChameleonMini #42: RevG no bootloader?

domoarigatosan, 2016-10-19:
Hi team,

I've received my RevG chameleon and proceed with the requisite steps before upgrade to new firmware from my Win7 64bit machine.

1) Installed DFU programmer. 
2) Installed Chameleon driver and able to see "ChameleonMini Virtual Serial Port [x]" from TeraTerm.
3) Perform self test and everything was passed from TeraTerm.
4) Hit upgrade from TeraTerm, hang from there. Red light from Chameleon was off.
5) Executed ChameleonFirmwareUpgrade.bat and shows error that chip is empty. Required to press any key to continue as there is no bootloader in the chip.

Performed the similar steps on a WinXp 32bit machine also giving the same result and in the same time shows error on line 7 on Chameleon-mini.eep file.

I believed that I have accidentally wiped the bootloader and now required to restore back to original state.

Please help, Thanks

geo-rg, 2016-10-24:
Hi @domoarigatosan,
first things first. Please connect your Chameleon to the PC while holding the right button (the outermost one). Then go to the device manager and check whether the Chameleon shows up as something like "DFU ATXMEGA128A4U". If so, you did not delete the bootloader. Then you could use another programmer instead of the .bat file. Therefor, please refer to https://github.com/emsec/ChameleonMini/issues/17#issuecomment-255729016.

If you accidentally wiped the bootloader, you can find it in this GitHub project (Firmware/Atmel DFU Bootloader). Unfortunately, you cannot flash this with USB but you would have to use an external programmer.

domoarigatosan, 2016-10-26:
@geo-rg I have tried your suggestion yet no luck :(

I have tried on winXP and win7 machine also can't flash RevG as per instructed. Below are the errors occurs on both machine. The RevG totally no lights even I have unplugged or plugged to USB.

**Using Atmel Flip on WinXP** (after pressed start application, nothing happened)
[image]

**Error starting RevG on WinXP**
[image]

**Line 7 error when executing ChameleonFirmwareUpdate.bat**
[image]
**Same error as well on Win7**
[image]

domoarigatosan, 2016-11-09:
Issue resolved, after using Zadig to reinstall the drivers and managed to upgrade the firmware from the .bat file

closed 2016-11-09 by domoarigatosan