emsec/ChameleonMini #46: Handle HALT properly

doegox, 2016-10-19:
Hi, I tried MF_ULTRALIGHT against a hotel door but it was highly unreliable.
The problem is that the emulation  doesn't handle the STATE_HALT properly.
It should wake up *only* when receiving a WUPA and not when receiving a REQA otherwise the reader is discovering the same card over and over.

I tested it with an ugly patch, works better. I may provide a proper patch when having a bit more time...

dev-zzo, 2016-10-19:
True. This implementation of ISO 14443-3 Type A cards seems to differ from the standard for both Classic and Ultralight code... at least in two important aspects: a) accepting REQA/WUPA in any state and b) not differentiating between IDLE and HALT states.

@david-oswald Could you let us know the reasons behind this divergence from the standard? Should we submit patches to rectify this?

geo-rg, 2016-10-19:
You could submit patches. A good approach would be to change the ISO14443AWakeUp function or splitting it up in two functions that check for WUPA and REQA seperately. Also, the states inside the card emulation applications have to be changed.

So: If you find time to change this to the standard right now, you could submit a patch. But we also have this on our agenda, although this may need more time than if you do it.

closed 2016-10-24 by geo-rg