emsec/ChameleonMini #51: Using the Chameleon as a Reader

uhillig, 2016-10-24:
I tried to use my Chameleon as a reader. I configured a slot in Reader mode and try to read in some cards I have. No matter which command I use I will soon ran into an Timeout. Even adjusting the timeout settings makes no difference.

I there something I've overlooked or is my Chameleon broken?

dev-zzo, 2016-10-24:
Please note the reader mode only supports ISO 14443 Type A compliant cards e.g. NXP MIFARE. If you happen to have cards of other types, you will get timeouts as you report. Please confirm the card types you are testing.

geo-rg, 2016-10-24:
Hi @uhillig,
can you describe what you did exactly?

grymoire, 2016-10-24:
So there's no support for 14443B cards?

geo-rg, 2016-10-24:
@grymoire 
The firmware does **not yet** support ISO14443B, neither as reader nor for emulation. The hardware supports both.

grymoire, 2016-10-24:
Okay - thanks. I hoped it was just a firmware issue, but I wasn't sure. I'd like to help if I can. We have a lot of HID cards in our area in the US.

uhillig, 2016-10-27:
I tried some NFC tokens which seems to be a Philips Mifare Standard 1k. These tags won't work. I doubled checked an tested the blank Mifare cards that came with my card reader. These cards are recognized by the Chameleon with no problem.

Thanks for your help.

closed 2016-10-27 by uhillig