emsec/ChameleonMini #52: chamtool.py in Win7 Simple Question How to get it to work from scratch.

zenroth1752, 2016-10-24:
I am trying to get the chamtool.py to work. I am new to python 
I have installed Pip and Pyserial manually on a win 7 machine. 
 
C:\Chemelon>pip list
pip (8.1.2)
pyserial (3.2.1)
setuptools (20.10.1)

What is the syntax of the argument that chamtool.py wants ? I have tried a number of combinations.
chamtool.py -p com6
chamtool.py - p 'com6' with quotes and without  with -p and with out. I will add my learning's as I am sure I can be the only person learning this from scratch.  I have it working with Xmodem but I just wanted to use the script. 

from pyserial "i have changed the id" in this text.
>>> ser
Serial<id=0x27dcf0, open=True>(port='com6', baudrate=19200, bytesize=8, parity='
N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False)

geo-rg, 2016-10-24:
Have you tried upper case letters, e.g. COM6?

zenroth1752, 2016-10-24:
I have just did tried COM6 and some variations.
This is the error:
c:\Chemelon\ChameleonMini-master\Software>chamtool -p 'COM6'
  File "C:\Chemelon\ChameleonMini-master\Software\chamtool.py", line 14
    print(formatString.format(timeString, text), file=sys.stderr)
                                                     ^
SyntaxError: invalid syntax

And if i call chamtool via Thonny (IDE) then I get this.
ImportError: No module named 'serial' Which is a bit more descriptive but  still the same root cause. I will do some more reading around pyserial and see what comes up.

zenroth1752, 2016-10-25:
This can be closed: -  See https://github.com/emsec/ChameleonMini/pull/53. Thank you https://github.com/doegox

closed 2016-10-25 by zenroth1752