emsec/ChameleonMini #58: Typo in doc

doegox, 2016-10-26:
Today we've "Driver" subdirectory in the repo

GettingStarted.txt:
  * "Drivers" subdir mentioned 3x

README.md:
  * Driver : The Chameleon driver for Windows

I suggest renaming the subdirectory "Driver" into "Drivers" as we've now bits for Win & Lin, and to fix README.md accordingly.
You may also consider serving 98-ChameleonMini.rules with rawgit as it's already done for the Windows drivers in GettingStarted

0x715C, 2016-10-26:
#60 Added link for 98-ChameleonMini.rules

doegox, 2016-10-27:
While at it, another one:

```diff
diff --git a/Doc/DoxygenPages/Reader.txt b/Doc/DoxygenPages/Reader.txt
index 5f53c1d..e258657 100644
@@ -64,8 +64,8 @@
  * 
  * \warning This command does not deactivate the reader field after finishing in order to make it possible to keep up the conversation with the card.
  * 
- * `SEND <2-BYTEVALUE> <BYTEVALUE>`
- * --------------------------------
+ * `SEND_RAW <2-BYTEVALUE> <BYTEVALUE>`
+ * ------------------------------------
```

geo-rg, 2016-11-21:
I merged right now and will rebuild the docs soon.

closed 2016-11-21 by geo-rg