emsec/ChameleonMini #62: Toolchain setup

solutionzone, 2016-10-26:
Hi folks,
Is there a tutorial or a link that anyone would be kind enough to share that explains how to set up the development environment for the firmware on a Windows based system?  I have quite a lot of embedded C experience, so that's not a problem, I have never developed for an Atmel device before, so i am keen to learn.  Also I have quite a lot of experience writing applications that use cards (in the transport industry), so that's not a problem.  I am keen to learn about the low level driving of cards now...

I have cygwin installed on my machine, and I downloaded the Atmel Studio v7.0.    When I run make in the Firmware directory, the first  error is "avr-size: command not found".  I guess there may be either a package missing from cygwin or the PATH is wrong.  C:\cygwin64\bin is in the path variable...

Any suggestions more than welcome.

Thank you
Neil

Xykon, 2016-10-27:
Did you add "C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin" to your PATH variable?

solutionzone, 2016-10-27:
That was it!  Thank you very much for your help.

closed 2016-10-27 by solutionzone