emsec/ChameleonMini #63: LED functionality disabled

SubversiveSecurity, 2016-10-27:
Following Upgrade the leds are deactivated, furthermore they are unable to be set. Steps followed as well as version information is listed bellow. 

Steps:
Connect to Chameleon using Minicom
version?                  - Returns version info bellow
ledgreen?                - Returns state of None
ledgreen=powered  - Returns 100 OK
store                        - Returns 100 OK 
ledgreen?                - Returns None
Disconnect from Chameleon and reconnect yields same result

OS : Ubuntu 16.04
MiniCom Version : 2.7

Version:
ChameleonMini RevG 161024 using LUFA 151115 compiled with AVR-GCC 4.9.2. Based on the open-source NFC tool ChameleonMini. https://github.com/emsec/ChameleonMini commit 2005f46

geo-rg, 2016-10-28:
Since I cannot reproduce this, I suppose you have forgotten to program the EEPROM. Are you sure that have written to EEPROM successfully during the firmware upgrade?

geo-rg, 2016-10-28:
I think I missed something when I read your issue. Not programming the EEPROM does not explain the behavior you are describing. To be clear: If ledgreen=powered return 100:OK and if ledgreen? returns NONE directly (i.e. no setting change in between) afterwards, there must be a firmware bug, which I cannot reproduce from here.

Thus, I suggest you try to flash the firmware again.

geo-rg, 2016-11-21:
@SubversiveSecurity Is your problem solved?

SubversiveSecurity, 2017-01-28:
Compiled firmware from source and issue was solved. Unclear what went wrong when updating firmware, if anything reveals itself I will make it know.

closed 2017-01-28 by SubversiveSecurity