emsec/ChameleonMini #64: Tips for Linux users

doegox, 2016-10-27:
As there is not really other place of discussion and my notes are probably not mature enough for inclusion in the official docs, I'm sharing a few tips here and hoping to gather yours as well!

I'm using a Debian, details may vary with other distros.

Setup
-------

First my `/etc/udev/rules.d/98-ChameleonMini.rules` to operate *and* flash the chameleon as regular user and stop being annoyed by pesky ModemManager:
```
# Rule for ChameleonMini RFID Research tool
ATTRS{product}=="Chameleon-Mini", SUBSYSTEMS=="usb", ATTRS{idVendor}=="16d0", ATTRS{idProduct}=="04b2", GROUP="users", MODE="0666", SYMLINK+="chameleon", ENV{ID_MM_DEVICE_IGNORE}="1"
# DFU mode
SUBSYSTEMS=="usb", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="2fde", GROUP="users", MODE="0666"
```

Minicom
------

As terminal, minicom can do the job, but a few tweaks help a bit, with this file as 
`/etc/minicom/minirc.chameleon` or `~/.minirc.chameleon`:
```
pu port             /dev/chameleon
pu localecho        Yes
pu linewrap         Yes
# disable other transports than xmodem
pu pname1           YUNYY
pu pname2           YUNYY
pu pname4           NDNYY
pu pname5           NDNYY
pu pname7           YUYNN
pu pname8           NDYNN
pu pname9           YUNYN
```
Then invoke it simply with:
```
minicom chameleon
```
xmodem transfers are possibles.

Screen
------
Usually I like `screen` as simple terminal but there is no easy way to activate local echo (see [this discussion](https://superuser.com/questions/752098/local-echo-using-screen-to-connect-to-a-serial-terminal))

Socat
------
So the alternative to get a simple terminal is to use socat. 
A nice add-on is to use its readline support to get history, ctrl-r etc. Unfortunately Debian and Ubuntu versions compiled it without readline support, so if you want to have it you've to recompile the package by yourself or use the original sources. 
But getting history in your chameleon terminal is so great!
```bash
#!/bin/bash
socat READLINE,history=$HOME/.chameleon_history /dev/chameleon,crlf
```

File transfer
------
I wish I could find a one-liner with e.g. `socat` & `rx` to get easy `upload.sh` and `download.sh` but so far my attempts of xmodem transfers failed. Anybody has better idea? 
Using `minicom` or `chamtool.py` are the alternatives.

Flashing
------
Here is my little script to compile, jump to bootloader, flash and jump back to normal mode in one go without having to unplug/plug the chameleon. This requires to install also the package `dfu-programmer`.
```bash
#!/bin/bash
cd ChameleonMini/Firmware/Chameleon-Mini
make || exit 1
# Jump to bootloader mode
echo -e "\rUPGRADE\r" > /dev/chameleon && sleep 1
make program
# Jump to normal mode
dfu-programmer atxmega128a4u start
```

Your turn
------
Any other tips to share on your side?

zenroth1752, 2016-10-27:
You are star - thank you

zenroth1752, 2017-02-06:
This tool allows the MFD dump to be in human readable format - https://github.com/zhovner/mfdread