emsec/ChameleonMini #67: Add multiline result support

doegox, 2016-10-28:
Issue #16 (VERSION returning several lines) was solved by simply writing all the info in a single line.
But `dump_mfu` returns also several lines and probably more commands will do in the future.
So what about adding a new response type for multiline result?
E.g.
```
102:OK WITH MULTILINE TEXT
```
that would e.g. terminate on a last line containing only `EOT` (end of transmission)

Example:
```
VERSION?
102:OK WITH MULTILINE TEXT
ChameleonMini RevG 161027 using LUFA 151115 compiled with AVR-GCC 4.9.2.
Based on the open-source NFC tool ChameleonMini.
https://github.com/emsec/ChameleonMini commit 70b6cce
EOT

DUMP_MFU
102:OK WITH MULTILINE TEXT
04575B8031BF02800C48000000000000
000000009632F4000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
EOT
```

dev-zzo, 2016-10-28:
This approach is bound to fail in case someone willing to send the `EOT` itself. I would suggest separating human-readable and machine-readable output, for example, by providing separate commands where there are such issues (`VERSION` is a good example). This is not as bad in case of `DUMP_MFU` as the output has some expected format, which can be easily parsed by both humans and machines.

skuep, 2016-10-28:
I would just replace the string "EOT" by 0x04 from the ASCII table ("End Of Transmission"), since all output from the chameleon is ASCII. A human readable terminal would probably don't display the EOT character.

geo-rg, 2017-04-05:
I added a pull request with a first draft of multiline response handling (see #105). I would suggest to use the pull request to discuss and solve the problem, but I also will leave this issue open until this thing is solved.