emsec/ChameleonMini #70: Chamtool.py didn't work?

ykpe, 2016-10-29:
My chamtool.py keep failing as issue#16.
  File "/multimedia/git/ChameleonMini/Software/Chameleon/Device.py", line 134, in writeCmd
    statusCode, statusText = status.split(":")
ValueError: need more than 1 value to unpack

Therefore, I've update firmware to ChameleonMini
But chamtool.py didn't react to -p comport(Mine was COM3)
[image]

System:Win 10 64-bit, Python ver 3.5

One more question, 
My dump_mfu always showed timeout when dumping my classic 1K card by using Teraterm
Reading works find for me, though. 

Any suggestion?

doegox, 2016-10-29:
`dump_mfu` is for MIFARE Ultralight cards, not MIFARE Classic

By default `chamtool.py` doesn't output anything, try instead
```
python chamtool.py -p /dev/chameleon -v
```

geo-rg, 2017-04-05:
@ykpe Did this solve your problem?

closed 2017-07-12 by geo-rg