emsec/ChameleonMini #73: Test  Buttons: FAILED

testtest666, 2016-11-01:
After the boot, the following error, how to solve?（Repeatedly restarted）

===================================
You can also use the following commands:
* Type "showlog" to show the factory test log
* Type "reset" to re-run the self test (you need to power cycle the Chameleon)

Generating Test Report from SRAM:
Test Report for Chameleon-Mini
* FW version: 0.5
* Test      Buttons: FAILED (Took 16584ms) Time exceeded without input on Left Button, time exceeded without input on Right Button.
* Test       Reader: FAILED (Took 529ms) Could not perform a whole ISO14443A select protocol and finally receive a SAK without errors. Used threshold range: 50 - 4095.
* Test         FRAM: PASSED (Took 149ms) Wrote 32768 pseudorandom bytes to FRAM. Successfully read the same bytes from FRAM.
* Test        Fuses: PASSED (Took 1ms) Read the following fuse bytes (the values in brackets have the form <reference value;mask>). 1: 00(00;FF) - 2: BE(BE;63) - 4: F3(F3;1F) - 5: ED(ED;3F).
* Test    Signature: PASSED (Took 0ms) Read the Device ID 1E9746, expected 1E9746.
* Test     USB Enum: PASSED (Took 137ms) USB Device State is ADDRESSED.
* Test          VCC: PASSED (Took 106ms) Measured a VCC of 3330 mV.
* Test      Antenna: PASSED (Took 107ms) Measured the following RSSI values: 12632 mV, 12503 mV, 12468 mV, 12562 mV, 12457 mV.
* Total Testing Time: 17614ms
Your Chameleon will emulate a card with the UID 3D03F2C9.
Sorry. Some test(s) have FAILED, please unplug chameleon, fix errors and re-test.
Welcome to your ChameleonMini! This is test firmware version 0.5.

Sorry. Some test(s) have FAILED, please unplug chameleon, fix errors and re-test.

Type "upgrade" to jump to the bootloader and install new firmware as described on https://github.com/emsec/chameleonmini

dev-zzo, 2016-11-01:
Have you tried the advice it gives you at the very end (flashing the production firmware)? Do buttons work then?

geo-rg, 2016-11-01:
@testtest666 Please tell us what exactly you did. You received your Chameleon and then plugged it in and typed `showlog` and received this test log? Or did you type `reset` and rebooted the Chameleon in between?

emsec, 2016-11-01:
the test firmware expects you to place a card on the antenna, then press both buttons, and then all tests will pass. however, you don't need the test firmware any more (that was for the factory) so please follow the getting started to program the current firmware.

testtest666, 2016-11-02:
Thank you all。

I have updated the firmware to
Version?
Based on the open-source NFC tool ChameleonMini. Https://github.com/emsec/ChameleonMini commit 2005f46

But I have not tested, I do not know if there are any problems.

Where can I download test firmware version 0.5 (has been compiled)?
I would like to test again. Thank you.

geo-rg, 2016-11-02:
There is no possibility to download it. The test firmware was designed so the factory can check whether there are problems with some of the components. You are not supposed to use the test firmware for testing, it was just for the factory.

If you received your Chameleon and after powering it, _only_ the green LED turned on, everything is fine and you can enjoy your Chameleon. If you do not remember, just try to use the Chameleon as a reader (config=iso14443a_reader) and try to use the buttons (type rbutton=? for help).

testtest666, 2016-11-02:
in my impression：
The first time you use Tera Term to connect to Chameleon-mini, all tests are OK.
Once, I test firmware version 0.5,
executed the  “upgrade”  command, nothing has been done. Because there are other things, I left.
come back, I restart the Chameleon-mini. There  Test Buttons: FAILED.

geo-rg, 2016-11-02:
Ok, then everything is fine. I also noticed, that the test report you gave us in your first post states "Generating Test Report from SRAM". If this would have been the factory test, it would have been generated from FRAM. So what happened is that you restarted the test (I don't know how) and left it alone. Thus, the button test and the reader test failed.

Enjoy your Chameleon!

closed 2016-11-02 by geo-rg