emsec/ChameleonMini #76: Implement unit test builds

dev-zzo, 2016-11-06:
Considering how large the firmware will soon grow and how easy (hehe...) it is to break things when making changes to some internal functions, I believe it will be *quite* beneficial to make a new build target that would make a "test build". This special build would look quite like the factory-burned test firmware you already ship, but with many more tests added for the various internal functions. I have at least the following candidates:

* Various checksums and crypto: ISO 14443 CRC-A, Crypto1, TDEA, AES, and others
* ISO 14443-3 and/or 14443-4 parsers and state machines

What do you think? This may not test it all (e.g. codecs will be challenging to test in that way), but still may reduce the required testing effort considerably.

zenroth1752, 2016-11-11:
Great idea

geo-rg, 2016-11-21:
Sorry for the late reply. It really is a great idea! However, it is not at our short-term agenda. It would be awesome if anybody could do this!