emsec/ChameleonMini #78: Support for ATXmega32a4u MCU

auramirror, 2016-11-13:
Hi,
I just bought a ChameleonMini with ATXmega32a4u MCU. It's running the firmware rev E. I tried to upgrade to revG but I can't due to limited memory of the Atmel. 
 1) Is there any revision between E and G that may be supported in the mega32a4u MCU? 
 2) What important functions am I missing in rev E?
 3) Are there any plans in supporting the ATXmega32a4u in the future?

Thanks!
Mel

feifeilanniao, 2016-11-13:
where can you buy a chameleonmini with mega32a4u?

auramirror, 2016-11-13:
Here: https://store.ryscc.com/products/chameleonmini

Looking forward for some answers.

Thanks!

emsec, 2016-11-13:
Hi auramirror, I wish that I never had to explain this (see schematics and description, it makes everyting clear...) but it is like this: of course it is impossible to flash the RevG firmware to a RevE chameleon... not only the microcontroller but also various other stuff on the PCB has changed --> the firmwares are competely incompatible and you will probably destroy your RevE when uploading a RevG Firmware. The last state of RevE firmware is in the "RevE" folder in the github project - please use this for your outdated Chameleon :)

closed 2016-11-13 by emsec