emsec/ChameleonMini #79: My chameleonmini REV_G can't emulate the MF_CLASSIC_1K

feifeilanniao, 2016-11-14:
When i try to read my chameleonmini with my Mobile Phone,some problem occur.It seem that the chameleonmini can't communicate with the Phone stably.But when i CONFIG the chameleonmini as MF_ULTRALIGHT,it  perform well.
[image]
[image]

feifeilanniao, 2016-11-14:
[image]
emulate the MF_ULTRALIGHT

closed 2016-11-15 by feifeilanniao

peter-neu, 2016-11-16:
Had the same problem, what was the solution to this?

feifeilanniao, 2016-11-16:
Try to use the battery to supply power.

emsec, 2016-11-25:
sure that you use the correct secret keys? even when NFC phones have a bad read range (and most of them have a VERY bad read range :) ) there must be some position in which it works. I assume you're using the latest firmware. Otherwise, we need to find out what these Smartphones do different when reading out cards, compared to "normal" readers...  have no idea, yet. But thanks to everyone comtributing to this topic.