emsec/ChameleonMini #80: how to compute the key ?

feifeilanniao, 2016-11-17:
[image]
I config the chameleon as MF_CLASS_1K,and put it to the reader.Then i upload the log as you can see in the picture.With the uid,nT,nR,aR,how to compute the KEY?

geo-rg, 2016-11-21:
If you emulate a MiFare Classic card, you can simply download the cards contents via `download`. In this dump, the keys can be found at byte positions 48 and 58 (I'm not totally sure about the positions, but the keys are there somewhere).

feifeilanniao, 2016-11-21:
Sorry for my statement,which make some misleading.Also,thanks for your response.And for my question"how to compute the KEY",i want to use chameleonmini(config as MF_CLASS_1K) to sponf the reader,then i get the communication messages betweeen the chameleonmini and the reader,then with these messages,i want to compute the key that the reader  had used for authing.

geo-rg, 2016-11-21:
Ah ok. So afaik nT is plain and thus does not depend on the key. nR and aR are encrypted. The most recent and also best attack on MiFare Classic I know is from last year and uses several thousands of nonces (Meijer, Verdult at ACM CCS). Thus, I don't think that it is possible to compute the key from only one communication.

feifeilanniao, 2016-11-21:
@geo-rg 
But i know that someone can use chameleonmini to sponf the reader and get the key in minutes.How can they do this?Are there any methods or tools?

dev-zzo, 2016-11-28:
@feifeilanniao Use Proxmark3 for key recovery attacks on (older) MIFARE Classic cards (see the `hf mf mifare` command in the proxmark3 client tool) -- this one is tested and working. ChameleonMini does not have any attacks implemented; you are very welcome to contribute your working implementation of the existing attacks though. :-)

geo-rg, 2016-11-28:
@feifeilanniao I found the attack you think about in a paper from 2008. Just search for "Dismantling Mifare Classic" on google scholar. At the end of the paper, there are attacks described.

geo-rg, 2017-03-12:
@feifeilanniao Did that answer your question?

closed 2017-04-05 by geo-rg

feifeilanniao, 2017-04-15:
Thanks for your answer.I didn't play with chameleonmin for a long time.Recently,i have read the paper about "Dismantling Mifare Classic",it's to difficult for me to get some useful message from the paper.I will try to review it and analyze.

david-oswald, 2017-04-16:
You can also look at the newer attacks that are available: http://www.cs.ru.nl/~rverdult/Ciphertext-only_Cryptanalysis_on_Hardened_Mifare_Classic_Cards-CCS_2015.pdf

feifeilanniao, 2017-04-16:
ok,thank you so much!