emsec/ChameleonMini #82: CR: block read/Write command within of the emulated MyFare Classic memory?

AndreasBujok, 2016-11-22:
I need a mode to change bytes (read/write) within of the emulated MyFare Classic memory.
Any planing for these type of feature?

Thanks
Andreas

geo-rg, 2016-11-28:
If you mean something like a command "set_byte <offset> <value>" which changes a specific byte to a specific value: No, this is not planned currently although it is not that hard to implement: You may start with a look at the functions MemoryWriteBlock/MemoryReadBlock in Memory.c. Adding a new command can be done in Terminal/{Commands.h,CommandLine.c,Commands.c}, just have a look how the other functions are defined (your task would need ExecParamFunc, look for COMMAND_SEND).

AndreasBujok, 2016-11-28:
Great

thanks for the answer 😊

BR Andreas

closed 2016-11-28 by AndreasBujok