emsec/ChameleonMini #84: ISO 13693 dump & emulate

pruwait, 2016-11-28:
I deed emulate card like this:
_EM Microelectronic-Marin SA EM4x3x SKIDATA keycard 
208 bytes ‣ 52 blocks, with 4 bytes per block 
Supported read commands: ‣ Single Block Read ‣ Multiple Block Read ‣ Get Multiple Block Security Status ‣ Get System Information AFI supported 
DSFID supported 
IC reference value: 0x02 
Customer ID: 0x066
ISO/IEC 15693-3 compatible 
ISO/IEC 15693-2 compatible 
Tag description: ‣ TAG: Tech [android.nfc.tech.NfcV, android.nfc.tech.NdefFormatable] android.nfc.tech.NdefFormatable android.nfc.tech.NfcV ‣ 
Maximum transceive length: 253 bytes_ 

Emulation must support UID and 52 blocks memory.
block      HEX data

0          03 08 E1 69
1          82 18 22 00
2          00 38 00 10
3          1C 48 33 00
4          1B 00 00 00
5          00 00 00 00
6          00 00 00 00
7          00 00 00 00
8          00 00 00 00
9          00 00 00 00
10         00 00 00 00
11         00 00 00 00
12         00 00 00 00
13         00 00 00 00
14         00 00 00 00
15         00 00 00 00
16         00 00 00 00
17         00 00 00 00
18         00 00 00 00
19         00 00 00 00
20         00 00 00 00
21         00 00 00 00
22         00 00 00 00
23         00 00 00 00
24         00 00 00 00
25         00 00 00 00
26         00 00 00 00
27         00 00 00 00
28         2A 80 53 42
29         1F 90 53 42
30         33 00 00 00
31         00 00 00 00
32         00 00 00 00
33         00 00 00 00
34         00 00 00 00
35         00 00 00 00
36         00 00 00 00
37         00 00 00 00
38         00 00 00 00
39         00 00 00 00
40         00 00 00 00
41         00 00 00 00
42         1C 11 5D 1E
43         C0 05 1B 01
44         A9 1D 1F 13
45         22 5D A4 7D
46         20 50 4C E5
47         1F 00 00 00
48         00 00 40 79
49         00 08 58 15
50         00 00 00 00
51         00 00 00 00 

Can ChameleonMini do it now?

DrWummi, 2016-11-28:
See #12

No 15693 yet.

dev-zzo, 2016-11-28:
I think this can be closed as a duplicate of #12.

closed 2016-11-28 by geo-rg