emsec/ChameleonMini #88: Bootloader gone

hdpro4u, 2016-12-24:
I seem to have botched the bootloader setup, and now i cannot get the .hex back onto the board.  how do i start over from scratch?  Right now, when plugged into USB, i dont get the Chameleon driver, only the ATMEGA in device manager.  

.hex on github only says not valid in FLIP.

hdpro4u, 2016-12-24:
little more update....
When i plug in chameleon, i get "ATMEL USB DEVICES" 'ATxmega128a4u' in device manager.
i cannot update to chameleon firmware using the .bat file.  i cannot get out of bootloader mode.

when i run the .bat file, i get 
"Checking memory from 0x0 to 0x1FFFF... Emplty.
Chip already blank, to force erase use --force.
Error Reading Line 1.
there was an error with executing this command.  Maybe your chameleon is not in bootloader mode?"

Thanks.

geo-rg, 2017-01-02:
Since no one else has the problem of a broken hex file, I assume that the current hex file is not broken. (To be sure, you could clone the repository and compile it on your own.)

Usually, it should not be possible to delete or alter the bootloader unless you used an external programmer. Is your Chameleon powered by battery while flashing the Firmware? This could cause problems, so please turn off the battery power! Also, only flash the firmware after pressing the right button on power-up.

hdpro4u, 2017-01-05:
Maybe I'm confused as to how to load the .hex. I have done something that yields no lights on the board at all, I assume it is in boot loader mode since it shows up as a device in dev manager. FLIP doesn't recognize it as a valid hex downloaded right off the repository. 

Unplugging and powering on via battery with known good voltage yields no lights either.

hdpro4u, 2017-01-07:
How do i go about erasing everything on the chip and starting over with the boot loader and chameleon hex?  in AVR Stuido i can read the chip, but still says it cannot enter boot loader mode.  holding the RBUTTON does not do anything when powering up.  it still loads in windows device manager as an ATMEL Device "ATxMega128A4U".

domoarigatosan, 2017-01-26:
@hdpro4u 

I had similar issue initially, till I have found out the solution. Your bootloader is not erase/empty. It's the upgrade.bat unable to identify the bootloader. 

Try check on my step-by-step at here:
https://github.com/emsec/ChameleonMini/issues/17

Try activate the ChameleonFirmwareUpgrade.bat (Doesn't matter if you can't see the serial box on TeraTerm).

Now your Chameleon RevG should be upgrading mode.
Once done, close the ChameleonFirmwareUpgrade.bat
Go to device manager > install driver > choose "have disk" > Select "ChameleonDriver.inf"
Try communicate from TeraTerm again.

geo-rg, 2017-02-13:
@hdpro4u Does this solve your problem?

Peterthegreat, 2017-03-04:
@hdpro4u Are you sure you downloaded the file correctly? My guess is that you right clicked and selected the web link to "save as" instead of the file. You must first click on the link, then you right-click on RAW and select save as.

http://i.imgur.com/B0Y9kD7.png

hdpro4u, 2017-03-06:
Peterthegreat is the GREATEST!  That was it... i was saving as at the level of the initial link, and not clicking raw.  followed the process again, and now have the green light!  good work team.

closed 2017-03-06 by hdpro4u