emsec/ChameleonMini #89: FRAM cleared on Boot?

hsanjuan, 2016-12-25:
This post mentions that FRAM might be cleared on boot despite it being supposed to be permanent:

https://ebfe.pw/chameleon-mini-with-macosx-101/

From my limited playing-around, this seems to happen. If the card is switched off and on, ./chamlog.py log download comes back empty. If the card is plugged while left on, it works.

geo-rg, 2017-01-02:
The Log is written to SRAM, which is cleared on power-down. You can use logstore to store the log from SRAM to FRAM (please notice the warning from [this page](https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/Page_CommandLine.html)).

Another thing is, if the data of the current setting is changed (e.g. during a Mifare Classic Write CMD), it is changed in FRAM. If the FRAM content is not stored back to Flash (store or setting change), the Firmware **will overwrite the FRAM memory block, where the current setting content is stored, on power-up with the content from Flash**! Thus, it could look like the FRAM is "cleared" although it is overwritten.

karmathub, 2017-01-23:
same here, if the card is switched off the FRAM be cleared.

geo-rg, 2017-02-13:
@karmathub Do you really mean that FRAM is CLEARED (i.e. content is 0x0000...00) or is it reset to the previous state. Then my explanation from above would fit.

karmathub, 2017-02-15:
well, I tested this:

---------minicom commands--------------------
config = ISO14443A_SNIFF
logclear
100:OK
logmem?
101:OK WITH TEXT
2048
------------------------------------------------------

===>[I sniffed a mifare 1k with my phone]

---------minicom commands--------------------
logmem?
101:OK WITH TEXT
2028
LOGSTORE
100:OK
------------------------------------------------------

---------shell commands------------------------
user@host:~/ChameleonMini/Software$ ./chamlog.py -m memory -p /dev/chameleon
53491 ms <+53491 ms>:CODEC RX         (4   bytes) [67ff235d]
53493 ms <    +2 ms>:CODEC RX         (8   bytes) [5fb2883809b237a2]

user@host:~/ChameleonMini/Software$ 
------------------------------------------------------

===>[with selector to ON: detached and reattached Chameleon mini from usb] next:

---------shell commands------------------------
user@host:~/ChameleonMini/Software$ ./chamlog.py -m memory -p /dev/chameleon
53491 ms <+53491 ms>:CODEC RX         (4   bytes) [67ff235d]
53493 ms <    +2 ms>:CODEC RX         (8   bytes) [5fb2883809b237a2]

user@host:~/ChameleonMini/Software$ 
------------------------------------------------------

(data are present in FRAM)

===>[detached and reattached with selector to OFF]

---------shell commands------------------------
user@host:~/ChameleonMini/Software$ ./chamlog.py -m memory -p /dev/chameleon

user@host:~/ChameleonMini/Software$ 
------------------------------------------------------

(data lost)

well. if I operate always with selector to ON the data will be retained, but if I turn it OFF data will be lost (cleared?)

Lot of thanks for your work.

geo-rg, 2017-02-16:
@karmathub I found a problem with the code, which causes this behavior. Actually, the FRAM is not cleared. But what happens is: `logstore` writes the log from SRAM to FRAM. Also there is a variable in **SRAM**, which stores the current write-position in FRAM. If you restart the Chameleon, a log init function is called, which sets this variable automatically to the beginning of the FRAM space for log. Thus, the data are there, but the firmware does not know that they are there.

A solution could be to write the current write-position also to FRAM and to get it from there on restart. But then there is the problem, that the firmware cannot know, whether this is a valid address or it is just random data. While typing this, I think I remember that I came across this problem when I wrote the code and decided that the FRAM in this case only serves as extra memory.

So maybe anybody has another idea how to solve this, so the Chameleon remembers a FRAM-stored log?

**To make it clear**: These thoughts apply only to the log store function. My explanation from above is still valid for the setting contents.

karmathub, 2017-02-16:
@geo-rg can you dedicate 5 byte (or 8) at fixed (initial) FRAM address to initialize at "0x00000" when call up the command: <LOGMODE=MEMORY> ?

geo-rg, 2017-02-21:
Our idea is to set a flag in EEPROM when the Firmware is flashed. On every chameleon startup we check this flag and if it is set, we know that the address in FRAM is not valid, so we set it to the standard value and clear the flag. If the flag is already cleared, we know the address is valid.

**It would be nice, if someone could hand in a patch.**

karmathub, 2017-02-23:
I hope to aiding in some way, but I'm not a developer....
anyway I'm reading your code with the hope to knowing it

geo-rg, 2017-04-05:
@karmathub @hsanjuan With the latest commits, your problems should not occur anymore. The Log now is automatically stored to FRAM and after rebooting the Chameleon, it is recognized.

I will close this issue. If you find out that your problem still is there, please re-open this issue.

closed 2017-04-05 by geo-rg