emsec/ChameleonMini #91: File .hex does not work

EnriqueSch, 2017-01-17:
Helo:

I download the last version of RevE for my Chameleon-mini, I tried to write the new firmware using the following sentence:
avrdude -c avrispmkii -p x32a4u -P usb -e -U flash:w:Chameleon-Mini.hex 
It worked OK,:

avrdude: 24444 bytes of flash verified
avrdude done.  Thank you.

But now, the board remain with the green led off, and my reader (omnikey) are not able to detect the ChameleonMini., but the board is alive, if I ask it for version I got:

101:OK WITH TEXT
Chameleon-Mini 150304 using LUFA 130901 compiled with AVR-GCC 4.8.1

Could you Pls. tell me what I`m doing wrong?

Benoit37000, 2017-01-17:
hello,
i thinks you need to flash eeprom file the .eep file

geo-rg, 2017-01-18:
Hi @EnriqueSch 
as already said, you need to flash the eep file, too. In your case you can do this with `avrdude -c avrispmkii -p x32a4u -P usb -e -U flash:w:Chameleon-Mini.hex -U eeprom:w:Chameleon-Mini.eep`.

Does this solve your problem?

EnriqueSch, 2017-01-18:
Hello:

 

No, this procedure did not solve it, the reader still is not able to “see” the card, the only difference now is that the green led is on, before (when I download just the .hex file) the green led was off.

 

I have a second Chameleon-mini card untouched, is there any way to download the firmware of that card (to a file), in order to compare both firmwares?.

 

Regards

Enrique

Enviado el: miércoles, 18 de enero de 2017 4:20
Para: emsec/ChameleonMini
CC: EnriqueSch; Mention

geo-rg, 2017-01-18:
Hi @EnriqueSch 
have you set the Chameleons configuration to e.g. Mifare Classic? This can be done by connecting the Chameleon to your PC via USB. Then use a terminal program like TeraTerm or minicom and connect to the Chameleons terminal. Then type config=MF_CLASSIC_1k and hit enter.

You may also read the RevG documentation (https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/Page_Configurations.html). The principle is the same.

EnriqueSch, 2017-01-18:
It worked… thanks a lot

 

Enrique

Enviado el: miércoles, 18 de enero de 2017 9:26

closed 2017-01-19 by geo-rg