emsec/ChameleonMini #92: Wiki: Upgrade Procedure for Windows Users

AndreasBujok, 2017-01-24:
Hi,

i've tested the rules for "Upgrade the Firmware" as described.
But i got the following error "The INF file you selected does not support this method of installation".
I'm using Win7/x64.

BR Andreas

AndreasBujok, 2017-02-07:
I've the solution after non reaction of these issue :-(

The given documentation is wrong or deprecated.
See the chapter 6.2.2 on the original Atmel sheet http://www.atmel.com/images/doc8429.pdf.

Thanks for the none support *grrr*

doegox, 2017-02-07:
If something has to be fixed in Chameleon Mini documentation, it would be better to leave the issue open...

emsec, 2017-02-08:
Philippe you're so right as usual :D  Georg is ill with fever since 2 weeks.... so we will catch up when he's back

doegox, 2017-02-08:
@emsec could you reopen the issue by yourself? I don't have privileges to do so.

AndreasBujok, 2017-02-08:
I do it self.

BR Andreas

geo-rg, 2017-02-13:
@AndreasBujok I'm really sorry. Usually I am responsible for answering issues, but by the time you wrote this issue I got the flu.

I added a _note_ right after the driver update procedure in the Getting Started page.

closed 2017-02-13 by geo-rg