emsec/ChameleonMini #94: chameleonmini setup  help

kanderson851, 2017-01-27:
hello mates !! i purchase the device and follow the step by step the setup quide without success. is any good person who can have time and help me to chameleonmini work on my computer? please contact me on my email kanderson851@mail.com  or leave any id i will contact you. good luck !

domoarigatosan, 2017-01-28:
@kanderson851 What's the error message did you observed? 
What are the steps did you performed? 

Please provide more info so we can help out.

The setup is very straight forward.

kanderson851, 2017-01-28:
thanks for help, i think i complete the start setup now is work fine is connecting to my computer, im on step 2 :)  try to read , upload and learn commands i will still need help do you able to assist me and maybe to connect on my pc with team viewer ? im able to pay for your time. plz email me. good luck!

zenroth1752, 2017-01-29:
Checking memory from 0x0 to 0x1FFFF...  Empty.
Chip already blank, to force erase use --force.
Error reading line 1.
There was an error with executing this command. Maybe your ChameleonMini is not
in bootloader mode?

I did have a problem with libusb-win32 or I installed from libusb-win32 is a port of libusb-0.1 under Windows 

And now Nothing is happening -- Oh no :-(

zenroth1752, 2017-01-29:
OK, so i have a backup Chameleon and it works fine. When I plug in the original Chameleon there are no lights and the last message I got on  attempting to upload the new firmware was "Checking memory from 0x0 to 0x1FFFF... Empty.
Chip already blank, to force erase use --force.
Error reading line 1.
There was an error with executing this command. Maybe your ChameleonMini is not
in bootloader mode?" and now no sign of life? In a word, Help! :-)

zenroth1752, 2017-01-29:
Mine is related to Bootloader gone #88. So I will close my bit

closed 2017-04-05 by geo-rg