emsec/ChameleonMini #98: OS X - Windows - No software connection with Chameleon

nozriv, 2017-02-15:
Hello,
I did the upgrade software with no problem on my Mac.
After what I tried to use CardPeek on my Mac and on my PC.
The Chameleon mini isn't recognized on my Mac or PC.
info from CardPeek:
0016 DEBUG   Scripts are up to date.
0017 DEBUG   Found 0 PCSC readers

I try Tera Term, same problem. Chameleon is not seen.
My friend who also have a Chameleon mini is stuck at the same point.
What did we miss? I did lots's of search on Google, but no answer found.
Any help will be appreciated.

Regards, 
Nico

nozriv, 2017-02-15:
I add this info for a better understanding:

> Mac-2:Chameleon nico$ sudo avrdude -c flip2 -p ATXMega128A4U -B 60 -P usb -U application:w:Chameleon-Mini.hex:i -U eeprom:w:Chameleon-Mini.eep:i
Password:

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9746 (probably x128a4u)
avrdude: NOTE: "application" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "Chameleon-Mini.hex"
avrdude: writing application (131072 bytes):

Writing | ################################################## | 100% 1.59s

avrdude: 131072 bytes of application written
avrdude: verifying application memory against Chameleon-Mini.hex:
avrdude: load data application data from input file Chameleon-Mini.hex:
avrdude: input file Chameleon-Mini.hex contains 131072 bytes
avrdude: reading on-chip application data:

Reading | ################################################## | 100% 0.67s

avrdude: verifying ...
avrdude: 131072 bytes of application verified
avrdude: reading input file "Chameleon-Mini.eep"
avrdude: writing eeprom (83 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 83 bytes of eeprom written
avrdude: verifying eeprom memory against Chameleon-Mini.eep:
avrdude: load data eeprom data from input file Chameleon-Mini.eep:
avrdude: input file Chameleon-Mini.eep contains 83 bytes
avrdude: reading on-chip eeprom data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 83 bytes of eeprom verified

avrdude done.  Thank you.
________________________
Mac-2:Chameleon nico$ nfc-list
nfc-list uses libnfc 1.7.1
No NFC device found.
_______________
Mac-2:Chameleon nicolasvirzo$ system_profiler SPUSBDataType
USB:

    USB 3.0 SuperSpeed Bus:
......................
    USB 3.0 Hi-Speed Bus:

      Host Controller Location: Built-in USB
      Host Controller Driver: AppleUSBXHCI
      PCI Device ID: 0x1e31 
      PCI Revision ID: 0x0004 
      PCI Vendor ID: 0x8086 
      Bus Number: 0x0a 

        Chameleon-Mini:

          Product ID: 0x04b2
          Vendor ID: 0x16d0
          Version:  0.01
          Serial Number: 537533334303FFFF71FFC100B000
          Speed: Up to 12 Mb/sec
          Manufacturer: Kasper & Oswald GmbH
          Location ID: 0x14100000 / 3
          Current Available (mA): 500
          Current Required (mA): 100
.........

geo-rg, 2017-04-05:
Hi @nozriv,
if I understand correctly, you tried to use the Chameleon as a reader with CardPeek? The Chameleon itself is no PC/SC reader.

However, connection with TeraTerm should work. What specific steps have you done in order to connect to the Chameleon with TeraTerm? Speaking for Windows, you could first identify the com port where your Chameleon is connected by opening the device manager and find the entry "ChameleonMini Virtual Serial Port" (or similar).

geo-rg, 2017-07-12:
Hi @nozriv,
is your problem solved?

geo-rg, 2017-07-26:
I assume the problem is solved, please simply re-open the issue if not so.

closed 2017-07-26 by geo-rg