emsec/ChameleonMini #100: problems with ChameleonMini RevF in sniff mode (card responses don´t seem to be logged correctly)

am-tim, 2017-02-20:
Hello everybody,

I am using a ChameleonMini RevF with the newest firmware (RevG) within a student project.
What I want to do is sniffing out the communication between a terminal and a MIFARE DESFire.
For this reason I am running the chamlog python script with python 3.6 on my windows 10 os.

This is what is logged during the initialization/ anticollision:

28334 ms <    +1 ms>:CODEC RX         (1   bytes) [26]
28334 ms <    +0 ms>:CODEC RX         (2   bytes) [0d0f]
28334 ms <    +0 ms>:CODEC RX         (2   bytes) [9320]
28335 ms <    +1 ms>:CODEC RX         (4   bytes) [9b3f4c22]
28336 ms <    +1 ms>:CODEC RX         (9   bytes) [9370880476669c6442]
28336 ms <    +0 ms>:CODEC RX         (3   bytes) [d50000]
28337 ms <    +1 ms>:CODEC RX         (2   bytes) [9520]
28337 ms <    +0 ms>:CODEC RX         (4   bytes) [d008fd03]
28338 ms <    +1 ms>:CODEC RX         (9   bytes) [95703ad01e8074b1cd]
28339 ms <    +1 ms>:CODEC RX         (3   bytes) [ef8031]
28451 ms <  +112 ms>:CODEC RX         (2   bytes) [5000]

So, obviously the commands from the terminal (REQA; select cascade 1 and 2 and so on..) are recorded correctly. But what about the responses from the card? 
At the place of the 0d0f I would expect a 4403 due to the fact that it is a MIFARE DESFire. 
By the way, the ChameleonMini in Reader Mode, using the IDENTIFY command, detects the ATQA 4403. 

Another example is the log of the communication between a SCL3711 usb rfid reader and a MIFARE Ultralight:

46385 ms <+46385 ms>:CODEC RX         (1   bytes) [26]
46385 ms <    +0 ms>:CODEC RX         (2   bytes) [cd0f]
46386 ms <    +1 ms>:CODEC RX         (2   bytes) [9320]
46386 ms <    +0 ms>:CODEC RX         (4   bytes) [9b1f8083]
46388 ms <    +2 ms>:CODEC RX         (9   bytes) [93708804fb9bec000e]
46389 ms <    +1 ms>:CODEC RX         (3   bytes) [fd0001]
46390 ms <    +1 ms>:CODEC RX         (2   bytes) [9520]
46391 ms <    +1 ms>:CODEC RX         (4   bytes) [00577f13]
46393 ms <    +2 ms>:CODEC RX         (9   bytes) [9570ea0b498028e49d]
46393 ms <    +0 ms>:CODEC RX         (3   bytes) [7f800c]
46626 ms <  +233 ms>:CODEC RX         (4   bytes) [500057cd]

Same problem here. Only the commands from the terminal seem to be logged correctly. 

Maybe anyone has the same problems or can help me solving this? 

Thanks, Tim

geo-rg, 2017-02-21:
The sniff mode currently does not support the PICC->PCD direction. I will leave this issue open, rename it and tag it as enhancement request.

Peterthegreat, 2017-05-25:
[image]

On the kickstarter page, the sniff mode is supported for PICC->PCD direction... Is the code somewhere but is not implemented on rev G?

geo-rg, 2017-05-26:
@Peterthegreat What you can see there is the log for an emulation. Within emulation, both directions are logged. But for sniffing the Chameleons current firmware is not capable of sniffing the PICC->PCD direction although it should be possible.

gypsophlia, 2018-08-15:
#180 #181

geo-rg, 2018-09-12:
As @gypsophlia already has commented, he has handed in a patch to enable both directions for sniffing.

closed 2018-09-12 by geo-rg