emsec/ChameleonMini #99: Could you please give an instruction on how to compile your Firmware code in Linux

shinestarqiao, 2017-02-20:
Hello,
I have bought some ChameleonMini chips and I want to modify the firmware as I want.
But I am not familiar with Firmware Programming.
I have met undeclared error like this:

AntennaLevel.h:40:19: error: ‘ADC_CH_START_bm’ undeclared (first use in this function)
  ADCA.CH0.CTRL |= ADC_CH_START_bm;
                   ^
AntennaLevel.h:41:31: error: ‘ADC_CH_CHIF_bm’ undeclared (first use in this function)
  while( !(ADCA.CH0.INTFLAGS & ADC_CH_CHIF_bm) );

Could you please help me solving this problem?

Thanks a lot

Regards
Qiao

geo-rg, 2017-02-21:
As for ubuntu , `sudo apt-get install avr-libc binutils-avr gcc-avr avrdude `, then switching to the Firmware/Chameleon-Mini directory and then `make` should do it.

shinestarqiao, 2017-02-21:
Thanks a lot for your answer. I follow your instruction and the errors are still there.
I compile the code in virtual machine. Would the VM be the reason to cause this error?

Thanks

geo-rg, 2017-02-21:
I don't think your VM is the problem. This error sounds a lot like there is a missing include in the code, but I assume you are using the GitHub code without any changes. If so, I think the compiler assumes you are compiling for another chip than atxmega128a4u and thus does not know these values. But in the makefile there should be a line like `MCU = atxmega128a4u`.
`ADC_CH_START_bm` for me comes from `/usr/lib/avr/include/avr/iox128a4u.h`.

shinestarqiao, 2017-02-21:
There is no iox128a4u.h in my computer in the same path. 
But I can find this file in AVR GNU Toolchain which I download from website.
Would it be the problem of the version of AVR?

skuep, 2017-02-22:
What Toolchain and what Version are you using on what operating System? 
Sounds to me that you have a rather old Toolchain installed.

closed 2017-02-23 by geo-rg