emsec/ChameleonMini #104: Mifare Ultralight EV1 support

ogsts, 2017-03-28:
Is it possible to emulate Mifare Ultralight EV1 using Chameleon Mini ?

http://www.nxp.com/products/identification-and-security/mifare-ics/mifare-ultralight/mifare-ultralight-ev1-contactless-ticket-ic:MF0UL1101DUD?tab=Documentation_Tab

doegox, 2017-03-28:
I'm afraid not currently. Feel free to propose a patch ;)

geo-rg, 2017-04-05:
@ogsts If you feel confident with C, I would encourage you to start a development on Mifare Ultralight EV1 emulation. AFAIK it uses ISO14443 only with low data rates, so you can use the codecs, which are already on the Chameleon right now. The only thing you have to do is implementing the Application layer.

You could start with a look at Firmware/Chameleon-Mini/Application/MifareUltralight.c. There you can see, how the application for MifareUltralight is realized. Maybe you copy-paste it to another file, e.g. MfareUltralightEV1.c and then add the EV1 functionality. We would look forward to you contribution! I also think you would get help from others here.

ogsts, 2017-04-12:
Hi
I have created first version of Mifare Ultralight EV1 support on Chameleon-Mini.
I can read it using phone and NXP TagInfo application identifies Chameleon Mini as Mifare Ultralight EV1.
I have turned LOGMODE=memory and I see all expected ISO14443A commands REQA, ATQA, SELECT_CL1, etc. However, target reader where I would like to use emulated EV1 does not read it. 
And log contains nothing. So, I believe it is related to antenna matching. Any idea if/how this can be tuned ?

geo-rg, 2017-07-12:
Hi @ogsts 
sorry for the late response. Have you solved this? 

If not: the log shows really nothing? Not even receiving the REQA? Then I would think that the log function was deactivated at that time ;) But also, there has been a fix within the codec, but the bug which was fixed would not cause receiving nothing.

arne182, 2017-10-27:
Hi @ogsts 
I would also be interested in testing and help integrating this feature.

dev-zzo, 2017-11-16:
@ogsts Ping. :-) If you don't mind sharing your implementation as a pull request, I am sure it will be greatly appreciated by the community here.