emsec/ChameleonMini #106: 9 months later....

acidhax, 2017-04-14:
And there's still no way for me to use my Chameleon Mini like how they showed in the kickstarter video.

david-oswald, 2017-04-16:
Can you be a little more specific and say which feature you are missing?

acidhax, 2017-04-19:
Dump my NFC cards and use it on my Chameleon.

geo-rg, 2017-04-26:
Hi @acidhax,
which types of cards would you like to dump and emulate on the Chameleon? I know, in one of the videos it seemed like everything is ready and you can hold your Mifare DESFire / Classic / PLUS cards to the Chameleon and it will directly copy the contents. However, in the first video it is clearly stated, that you will have to know the keys and if not, you will have to crack the card. Currently, it is necessary to use a commercial reader and nfctools/libfreefare/etc. for many card types to dump.

Basically, the Chameleon is capable of reading out any ISO14443A card (if the keys are known), even those who have crypto on it - the firmware is just not yet implemented for this. The whole project has been advertised as open source and if you think the project lacks functionality, you may help implementing this functionality! We are a small company and small companies have not much money for open source projects as we don't really earn money with this.

I gladly will help you in any - for me possible - way to make the Chameleon more usable for you, but you will have to understand our restrictions as well.

closed 2017-11-23 by emsec