emsec/ChameleonMini #107: RF carrier, sub-carrier frequency tuning parameters

ogsts, 2017-04-24:
I observe very weak sensitivity between NFC reader and ChameleonMini while in Mifare Ultralight emulation mode. I can read/write successfully ChameleonMini using Android phone though. Same NFC reader works without issues with http://emutag.com. I assume some fine-grained RF carrier, sub-carrier frequencies.
Can this be done by adjusting Codec parameters ?

ogsts, 2017-04-24:
Looks like carrier frequency can be adjusted by changing Codec.h:
#define CODEC_CARRIER_FREQ          13560000
and sub-carrier 
#define ISO14443A_SUBCARRIER_DIVIDER    16

geo-rg, 2017-04-25:
Hi @ogsts 
the Chameleon actually uses the field generated by the reader. If you change ISO14443A_SUBCARRIER_DIVIDER, the subcarrier definitely will have the wrong frequency.

CODEC_CARRIER_FREQ is only used by the reader part of the Chameleon and thus has no effect on the emulation.

If you have trouble with emulation, this could have different reasons: At first, if your Android phone has a metallic case, the reading range is very limited. The Chameleon also tends to emulate not so good in short ranges and better in like 1cm distance to the reader. Both of the things you want to adjust are with 95 % prob not the cause of your errors.

What is the exact type of your NFC reader?

ogsts, 2017-04-25:
Hi @geo-rg

I want to use ChameleonMini with NFC reader which is out of my
control - I dont have any additional information about model.

What I know is that reader accepts Mifare Ultralight
card emulated by http://emutag.com if it is at the distance ~0-3 cm
from reader.

However, ChameleonMini has several advantages over http://emutag.com: 

a) fully open source firmware
b) possibility to use ChameleonMini as NFC reader (without use of phone)
   to clone Mifare Ultralight cards

> the Chameleon actually uses the field generated by the reader

I know that in emulation mode ChameleonMini uses reader field.
My assumption was that reader which I use ChameleonMini with
could use carrier frequency which is slightly off from
standard 13.56 MHz but ChameleonMini is sensitive only in
this standard frequency.

When I turn LOGMODE=memory I dont get anything in the log
after use with reader. Not even ATQA. So, I believe issue is
at RF level

geo-rg, 2017-04-26:
Hi @ogsts 
> My assumption was that reader which I use ChameleonMini with
could use carrier frequency which is slightly off from
standard 13.56 MHz but ChameleonMini is sensitive only in
this standard frequency.
I think I was not clear enough in my previous comment about this ;) What I meant is: the Chameleon really uses the frequency from your reader.  If it is slightly different, the Chameleon still should recognize the readers request.

OK, so you are doing `logmode=memory` and there is not even a REQA (0x26) or a WUPA (0x52) from the reader? Or does the Chameleon recognize one message (REQA/WUPA) and does not respond? An idea would be to set `logmode=memory` and `config=ISO14443A_SNIFF` and hold this to your reader. Usually, this should detect the readers requests. You could also set `ledred=field_detected` to see, if the Chameleon even recognizes a 13.56 MHz field.

ogsts, 2017-04-28:
Hi @geo-rg 
Led function LED_FIELD_DETECTED is really useful. I captured log using CONFIG_ISO14443A_SNIFF (attached). It contains 0x52 for couple of times but otherwise it looks pretty random.
[data.txt](https://github.com/emsec/ChameleonMini/files/964065/data.txt)

geo-rg, 2017-06-23:
Hi @ogsts 
last week we have made an update within the ISO 14443 A emulation codec, just try again with the current "latest" firmware. Does this solve your problem?

geo-rg, 2017-07-26:
I assume your problem is solved. Please simply re-open this issue if not so.

closed 2017-07-26 by geo-rg