emsec/ChameleonMini #110: Chameleon how sinffer mifare classic ?

maxben14, 2017-05-03:
How sniff mifare classic with chameleon mini ?  I want sniff and after get key crypto hack.

geo-rg, 2017-05-04:
Currently it is only possible to sniff the PCD->PICC direction. This can be done by setting `config=ISO14443A_SNIFF` and `logmode=memory`.

maxben14, 2017-05-04:
Thank you. Can you write me please how after calc key mifare from log sniff ?

geo-rg, 2017-05-04:
There are several attacks published, so I suggest you ask Google (or Google Scholar) or look at the Proxmark code.

maxben14, 2017-05-04:
I correctly understood that the chameleon log is the same as the log of the proximark?
I watch this https://www.blackhat.com/docs/sp-14/materials/arsenal/sp-14-Almeida-Hacking-MIFARE-Classic-Cards-Slides.pdf log proxmark. In chemeleon same as in this aticle ?

geo-rg, 2017-05-04:
No. As I already mentioned, the current firmware of the ChameleonMini is not capable of logging the Card->Reader direction. Thus, the log won't look the same as in these slides.

maxben14, 2017-05-04:
Ok, then you can use the chameleon log link to the article or explain how to get the key out. I understand the log of the proximark, how to get out there, I do not really understand how to get the key out of the chameleon.

maxben14, 2017-05-04:
I correctly understood that the chameleon will only intercept Nr and Ar from reader to card and don;t see Nt and At numbers ?

maxben14, 2017-05-04:
How do I understand right to restore the key me need to intercept 2 authorizations in this case? I can use the mfkey32 utility to restore the key from the chameleon log?

geo-rg, 2017-05-26:
Hi @maxben14 
I cannot simply explain you how to break a Mifare Classic card as this is illegal. I **can** give you the hint that there are academic publications about that topic. But then again you have to understand them by yourself.

Reading these publications you should keep in mind, that the Chameleon currently does not support sniffing the direction from card to reader.

maxben14, 2017-05-26:
Hi geo-rg.

I know all these attacks, but they require the nonce of Nt that the chameleon does not catch. I watched the Mfkey32 and mfkey64 tools, but both attachments require this nonces Nt.
I only see the possibility of a chameleon emulate a classic to generate the nonces Nt, Nr, Ar, At. But the chameleon badly emulates the mifare classic, tell me how to set it up for classic emulation.

I get errors during the emulation of the mifare classic.

geo-rg, 2017-05-26:
@maxben14 
check out the [MFClassic_patch branch](https://github.com/emsec/ChameleonMini/tree/MFClassic_patch) and recompile (`git checkout MFClassic_patch` and then run `make` in Firmware/Chameleon-Mini). This should be a better working MF Classic emulation.

maxben14, 2017-05-26:
Thanks, I'll try a new version of the chameleon.
Tell me, please, I correctly understood that in the case of a chameleon, the classic can be intercepted by the key, only by emulation, since it does not intercept the random numbers nt, which are required in all known attachments.

maxben14, 2017-05-26:
I watch branch Firmware/Chameleon-Mini 4 last month updated, i have this last version and have problem with emulation classic.

maxben14, 2017-05-28:
I try git checkout and make branch for your instruction but i get error compile when i get make.
[makefailed.log.txt](https://github.com/emsec/ChameleonMini/files/1034541/makefailed.log.txt)

Can help me build update for MF Classic emulation please.

geo-rg, 2017-07-12:
As I already responded to your other issue, please try the new firmware as there have been some bug fixes.

geo-rg, 2017-07-26:
I assume your problem is solved, please re-open this thread if you have again any related problems.

closed 2017-07-26 by geo-rg