emsec/ChameleonMini #112: Emulation Mifare Classic

maxben14, 2017-05-16:
I want to do a log when a chameleon emulates a mifare classic. I want to log the nonces Nt, Nr, Ar. Prompt please the command for logging nonces from the implementation of krupot1.

geo-rg, 2017-05-26:
As you can read at [this page](https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/Page_CommandLine.html) of our documentation, you may set `logmode=memory` and later get the log with `logdownload`. You will have to find the nonces by yourself from that log.

You may also use the chamlog tool from the Software folder of this repository.

maxben14, 2017-05-26:
I get errors during the emulation of the mifare classic.
The chameleon emulates ultralights well, but produces errors when emulating the mifare classic.

geo-rg, 2017-07-12:
@maxben14 Have you tried the last versions of the Firmware? We have fixed some bugs.

geo-rg, 2017-07-26:
I assume your problem is solved, just re-open this thread if you have any new related problems.

closed 2017-07-26 by geo-rg