emsec/ChameleonMini #117: Emulating DESFire 4k EV1?

decidedlygray, 2017-06-14:
Hello,

I was wondering if it is currently possible to emulate MIFARE DESFire 4k EV1 cards? I saw on the [home page](https://github.com/emsec/ChameleonMini/wiki):
`
Cards that the ChameleonMini can emulate in principle include: NXP Mifare Classic, Plus, Ultralight, Ultralight C, ntag, ICODE, DESfire / DESfire EV1
`

But it's not clear what "in principle" means, and I am not seeing any [Configuration](https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/Page_Configurations.html) in the official documentation for doing this.

This may be related to #31, though its not clear what functionality has been merged into ChameleonMini/master, or if that is one of the goals of that fork.

Thanks!

geo-rg, 2017-06-16:
Hi @decidedlygray 
"In principle" means that the hardware supports this, but for example the DESFire cards are not fully supported yet by the firmware. Maybe the code in the fork is enough for your use case, but if not, you may have to wait for it or help us implementing it!

Does this answer your question fully?

decidedlygray, 2017-06-16:
It does! Thanks!

closed 2017-06-16 by decidedlygray