emsec/ChameleonMini #119: TNP3xxx Application

Symon998, 2017-06-17:
Any possibility of TNP3xxx application compatibility?

doegox, 2017-06-18:
It's already compatible as such with normal MFC emulation.

Symon998, 2017-06-19:
Hi Philippe,

Thanks for the response on this, when configuring the card to MF_Classic_1K any dumps sent to the card fail when dump originally from TNP3xxx

Any help would be great thanks

Simon Buchanan

geo-rg, 2017-06-23:
Hi @Symon998 
recently there have been some fixes for Mifare Classic emulation, please try if these solve your problems.

Symon998, 2017-06-23:
Hi @geo-rg
Will take a look and update firmware tonight - will respond on outcome

Any specific details on what the fixes are that went in and when they were updated with MFC? As I had flashed with latest firmware available last week?

Thanks geo-rg

geo-rg, 2017-06-23:
The updates have been done Wednesday last week and this week. There was one update within the emulation codec and various updates within the Mifare Classic 4K emulation. See also [here](https://github.com/emsec/ChameleonMini/commits/master).

geo-rg, 2017-06-30:
@Symon998 Did the updates solve your problems?

geo-rg, 2017-07-26:
I assume the updates solved your problems. If not so, please simply re-open this issue.

closed 2017-07-26 by geo-rg