emsec/ChameleonMini #121: Emulation error

Peterthegreat, 2017-06-23:
Just tried the latest version on the master branch that supposed to contain the mifare1K patch, but seems that it still has some errors. Here is the log:

`00000 ms <    +0 ms>:BOOT             (0   bytes) []
19343 ms <+19343 ms>:CODEC RX         (1   bytes) [26]
19343 ms <    +0 ms>:CODEC TX         (2   bytes) [0400]
19345 ms <    +2 ms>:CODEC RX         (2   bytes) [9320]
19345 ms <    +0 ms>:CODEC TX         (5   bytes) [0030000030]
19348 ms <    +3 ms>:CODEC RX         (9   bytes) [93700030000030eda4]
19348 ms <    +0 ms>:CODEC TX         (3   bytes) [08b6dd]
19352 ms <    +4 ms>:CODEC RX         (4   bytes) [6004d13d]
19352 ms <    +0 ms>:APP AUTH         (2   bytes) [6004]
19353 ms <    +1 ms>:CODEC TX         (4   bytes) [591c5b5d]
19354 ms <    +1 ms>:CODEC RX         (1   bytes) [26]
19354 ms <    +0 ms>:CODEC TX         (2   bytes) [0400]
19357 ms <    +3 ms>:CODEC RX         (9   bytes) [93700030000030eda4]
19357 ms <    +0 ms>:CODEC TX         (3   bytes) [08b6dd]
19361 ms <    +4 ms>:CODEC RX         (4   bytes) [6004d13d]
19361 ms <    +0 ms>:APP AUTH         (2   bytes) [6004]
19361 ms <    +0 ms>:CODEC TX         (4   bytes) [d86176b1]
19365 ms <    +4 ms>:CODEC RX         (4   bytes) [6004d13d]
19365 ms <    +0 ms>:APP AUTHING      (4   bytes) [b3b15569]
19365 ms <    +0 ms>:APP AUTH FAILED  (4   bytes) [b3d5c45c]`

Also, there is no "ar" inside the log. Can somebody confirm this?

Peterthegreat, 2017-06-26:
I must mention that I uploaded the pre-compiled files, because they were added a few days ago. I will compile the source again and post an update

Peterthegreat, 2017-06-28:
Recompiled the source, and the result is similar. My phone's nfc dose not recognize an emulated tag. Here is the emulation log. 
[image]

There is something wrong with the latest build.

geo-rg, 2017-06-28:
Indeed, the reader does not respond to the Chameleon after sending the "cards" nonce and the Chameleon interprets the readers re-try as the reading continuing the authentication and everything goes wrong.
Which reader are you using?

Peterthegreat, 2017-06-28:
Right now, the reader from a public bus (first post) and the one from my phone is a Broadcom chip (second post). I will post the test from my PN532 later

geo-rg, 2017-06-30:
@Peterthegreat We cannot reproduce these errors. Is it possible that you use a self-built Chameleon? Then I would also take hardware errors into consideration.

Peterthegreat, 2017-06-30:
@geo-rg Yes. I am using a self-build Chameleon. In this post I tried the latest master branch, but using an older version of MFClassic_patch works perfectly fine with the hardware.

geo-rg, 2017-07-05:
Has your Fuse fix also solved this problem?

Peterthegreat, 2017-07-05:
Flashed the latest version. The log from PN532 looks good, but I think that worked even before the MFClassic_patch. I will try the reader that gave errors as soon as I have access to it.

Peterthegreat, 2017-07-05:
Although I was unable to store the log (even with Lbutton config set to store_log), the emulation finally worked (took 4 tries before it was recognized)

closed 2017-07-05 by Peterthegreat